STMicroelectronics/STM32CubeWL #1: problem compilling example from ST32cubeWL : Sigfox_AT_Slave

timFaivre, 2021-01-11:
Hi,
I tried to set up the cubeIDE to compile applications provided in the ST32cubeWL, in order to use them with the nucleo wl55jc1 evaluation board. 

The version in use are : 
STM32Cube_FW_WL_V1.0.0
 STM32CubeIDE_1.5.1 

Everything seems to work while compiling the LoRaWAN_AT_Slave applications (hence I suppose the overall configuration is correct.) 

When I try to compile the Sigfox_AT_Slave application, I got 26 errors, [build_error.txt](https://github.com/STMicroelectronics/STM32CubeWL/files/5797494/build_error.txt) 

 The first one being : 

```
arm-none-eabi-gcc -o "Sigfox_AT_Slave.elf" @"objects.list"  -l:Sgfx-Cmac-V1.0.0-CM0-O3.a -l:Sgfx-Monarch-V2.0.0-CM0-O3.a -l:SFX_LIB_V2.8.1_SE_FDL_MON.a -l:SFX_ADDON_RFP_V0.6.0_SE_FDL_MON.a -mcpu=cortex-m4 -T"C:\ST\tfa_prj\Sigfox_AT_Slave\STM32WL55JCIX_FLASH.ld" --specs=nosys.specs -Wl,-Map="Sigfox_AT_Slave.map" -Wl,--gc-sections -static -L../Middlewares/Third_Party/Sigfox/Crypto -L../Middlewares/Third_Party/Sigfox/Monarch -L../Middlewares/Third_Party/Sigfox/SigfoxLib -L../Middlewares/Third_Party/Sigfox/SigfoxLibTest --specs=nano.specs -mfloat-abi=soft -mthumb -Wl,--start-group -lc -lm -Wl,--end-group
../Middlewares/Third_Party/Sigfox/SigfoxLib\SFX_LIB_V2.8.1_SE_FDL_MON.a(sigfox_api.o): In function `SIGFOX_API_get_version':
D:\SVN\TST_U_LIB\SFX_LIBRARY\tags\V2.8.1\precompil\SE_FDL_MON\src\sigfox_api.c:(.text+0x262): undefined reference to `SE_API_get_version'
```

I did the setup using the File>new> stm32 from an ioc file option. 
Then try to build without touching any of the option...

(by the way, the issue template depicted [here](https://github.com/STMicroelectronics/STM32CubeWL/blob/747d4e27df07dcfc995b54f8af8a0bee38dfef8d/CONTRIBUTING.md) :  cannot be found for this project , so sorry if the format is not adequat...

FlorentVSTM, 2021-01-18:
Hi Timothé,

The libraries definition are not in the right order.
Please change the order through the project properties as follows:

[image]

Note a fix has been done in CubeIDE and will be available in the next release (v1.6.0)
Sorry for inconvenience.
Kind Regards

timFaivre, 2021-01-19:
Perfect, that's it ! 
Sorry if the question was obvious