STMicroelectronics/STM32CubeWL #2: LoRaWAN Class A LowPowerMode

karthirockz, 2021-01-26:
Hi,

We are using using https://github.com/STMicroelectronics/STM32CubeWL/tree/main/Projects/NUCLEO-WL55JC/Applications/LoRaWAN/LoRaWAN_End_Node to migrate our existing project from STM32L073, Its quite easy to enable LowPowerMode mode in https://github.com/Lora-net/LoRaMac-node/blob/2d5b3f05f31c47ff0f06530c51727ab7e5fe2405/src/apps/LoRaMac/classA/NucleoL073/main.c#L1215

Can you please let us know how to enable LowPowerMode in STM32WL

Thanks

ASELSTM, 2021-02-09:
Hi @karthirockz,

Allow me first to thank you for your contribution. Your question has been forwarded to our development team. We will get back to you as soon as we have more details.

With regards,

YoannBo, 2021-02-11:
Hi,

LowPower mode are active by default:
to change this, you can update LoRaWAN_End_Node\Core\Inc\sys_conf.h 

/**
  * @brief Enable Debugger mode
  * @note  1:ON it enables the debbugger plus 4 dgb pins, 0:OFF the debugger is OFF (lower consumption)
  */
#define DEBUGGER_ON       0

/**
  * @brief Disable Low Power mode
  * @note  0: LowPowerMode enabled. MCU enters stop2 mode, 1: LowPowerMode disabled. MCU enters sleep mode only
  */
#define LOW_POWER_DISABLE 0

bye

ASELSTM, 2021-03-12:
Hi @karthirockz ,

Please allow me to close this thread as no activity has been registered since a month. You can reopen it at any time if you have details to provide us to help you.

Thank you for your comprehension and thank you once more for your contribution

With regards,

closed 2021-03-12 by ASELSTM

YoannBo, 2021-06-10:
Hello @aessig ,

By using  LoRaWAN_End_Node from CubeWL, not touching anything, you should reach around 2uA (MCU+RTC+Radio in sleep mode) with the Nucleo

Best regards,