STMicroelectronics/STM32CubeWL #3: SIGFOX MONARCH timeout in Sigfox_AT_Slave

timFaivre, 2021-02-05:
**Caution**
The Issues are strictly limited for the reporting of problem encountered with the software provided in this project.
For any other problem related to the STM32 product, the performance, the hardware characteristics and boards, the tools, or the environment in general, please post a topic in the [ST Community/STM32 MCUs forum](https://community.st.com/s/group/0F90X000000AXsASAW/stm32-mcus).

**Describe the set-up**
 * The board (either ST RPN reference or your custom board).
NUCLEO WL55JC1
 * IDE or at least the compiler and its version.
STM32CubeIDE Version: 1.5.1Build: 9029_20201210_1234 (UTC)

**Describe the bug**
The timer during the monarch scan ends prematurly

**How To Reproduce**
1. Indicate the global behavior of your application project.
- compile and load Sigfox_AT_Slave
- connect to serial
- execute AT$MN=300 command (beacons are emitted every 300 seconds)
''' 
10s300:RF_API_init in MN20
10s301:MONARCH_TIM_START: 300 ms
10s302:MN Init
10s602:MN Deinit
10s602:MONARCH_TIM_STOP
10s602:RF_API_stop
NO RC found
'''
the timer is wrongly set to 300ms instead of 300s 

2. The modules that you suspect to be the cause of the problem (Driver, BSP, MW ...).
3. 
mn_api.c >> MN_API_TimerSart 
line 277 : UTIL_TIMER_SetPeriod(&Monarch_TimerTimeout, timer_value_ms);
the value passed to the function is in seconds while the timer expect ms
multplying the timer by 1000 fixes the issue

ASELSTM, 2021-02-09:
Hi @timFaivre,

Thank you for this other report. It has been forwarded to our development teams. We will get back to you as soon as they provide us with their feedback.

Best regards,

YoannBo, 2021-02-12:
Hello,

Thanks for your feedback,
i could indeed reproduce the issue. Please find below outcome of investigations:
This happens in _debug_ mode only. In _release_ mode, i find it OK.

The issue happens within `AT_scan_mn `function from **sigfox_at.c**
it is related to:

```
sfx_timer_unit_enum_t unit = SFX_TIME_S;
SIGFOX_MONARCH_API_execute_rc_scan(rc_capabilities_bit_mask, timer, unit, app_callback_handler);
```
in _debug_ mode, `unit `is not initialized properly. It is set to `SFX_TIME_MS`.. while in _release_ mode it is initialized properly.
I can't figure out why.

if you replace 

`SIGFOX_MONARCH_API_execute_rc_scan(rc_capabilities_bit_mask, timer, unit, app_callback_handler);`

by

`SIGFOX_MONARCH_API_execute_rc_scan(rc_capabilities_bit_mask, timer, SFX_TIME_S, app_callback_handler);`

this should be OK,

please let me know

timFaivre, 2021-02-23:
Hi,
sorry for the delay ! 

I agree with your conclusion, the fixe you proposed works correctly.

thanks for you tour time !

closed 2021-02-23 by timFaivre