STMicroelectronics/STM32CubeWL #4: Preamble Length in LoRaWAN_End_Node application

piyushy6, 2021-02-07:
Hi,
I am using WL55JC1 board with STM32CubeIDE. Can you please tell how to increase the Preamble Length in End Node application [https://github.com/STMicroelectronics/STM32CubeWL/tree/main/Projects/NUCLEO-WL55JC/Applications/LoRaWAN/LoRaWAN_End_Node] ?

ASELSTM, 2021-02-09:
Hi @piyushy6,

Allow me first to thank you for your contribution. Your question has been forwarded to our development team. We will get back to you as soon as we have more details.

With regards,

YoannBo, 2021-02-11:
Hi,

 The LoRaWAN_End_Node application is LoRaWan compliant project application.
Therefore, it follows the LoraWan Alliance specification in particular the Regional Parameter specification.
The Preamble format is defined in this document and hence its length is frozen.
For example for Europe it is defined as:
[image]

and is defined the same way for all regions

hope it answers your question.

piyushy6, 2021-02-18:
Thanks. Is there any example for LoRa to LoRa communication for stm32wl?

wolsty7, 2021-02-19:
I believe this example is what you are looking for: https://github.com/STMicroelectronics/STM32CubeWL/tree/main/Projects/NUCLEO-WL55JC/Applications/SubGHz_Phy/SubGHz_Phy_PingPong

piyushy6, 2021-02-19:
> I believe this example is what you are looking for: https://github.com/STMicroelectronics/STM32CubeWL/tree/main/Projects/NUCLEO-WL55JC/Applications/SubGHz_Phy/SubGHz_Phy_PingPong

piyushy6, 2021-02-19:
Hi, Thanks for sharing. 
This is specifically for Ping-Pong. 
Is there any way to convert this from Ping-Pong to fixed Receiver and Transmitter application?
What changes should I make to use one board as receiver and one as transmitter?

closed 2021-02-19 by piyushy6

wolsty7, 2021-02-21:
You could hard code one to always have the "isMaster" state as true and one as false, forcing one top be the transmitter and one the receiver.
https://github.com/STMicroelectronics/STM32CubeWL/blob/747d4e27df07dcfc995b54f8af8a0bee38dfef8d/Projects/NUCLEO-WL55JC/Applications/SubGHz_Phy/SubGHz_Phy_PingPong/SubGHz_Phy/App/subghz_phy_app.c#L252

piyushy6, 2021-02-22:
Thanks. Can you please also tell what does the function UTIL_SEQ_RegTask((1 << CFG_SEQ_Task_PingPong_Process), UTIL_SEQ_RFU, PingPong_Process) do?

https://github.com/STMicroelectronics/STM32CubeWL/blob/747d4e27df07dcfc995b54f8af8a0bee38dfef8d/Projects/NUCLEO-WL55JC/Applications/SubGHz_Phy/SubGHz_Phy_PingPong/SubGHz_Phy/App/subghz_phy_app.c#L229

YoannBo, 2021-02-22:
Hi,

UTIL_SEQ is the sequencer. It is not an OS but a 'run to completion' framework. (see 
`UTIL_SEQ_RegTask` registers a task. It basically 'associates' an ID (a bit map) to a function (here PingPong_Process)

Is counter part is `UTIL_SEQ_SetTask`. It will make the sequencer executing the registered function in the while(1) loop without any race condition. Therefore, the function can be executed in background and not in the interrupt context.

please find below a comparison between the 'standard way' and the 'sequencer way' doing exactly the same thing

[image]

Hope it answers your question

best regards

piyushy6, 2021-02-22:
Ok, thanks. 

This is my transmitter code, but I am unable to transmit anything. Can you please tell why?

[Tx.c.zip](https://github.com/STMicroelectronics/STM32CubeWL/files/6022086/Tx.c.zip)

YoannBo, 2021-02-22:
you are missing a Radio.Send(Buffer, BufferSize); line 79

piyushy6, 2021-02-22:
Thanks a lot. It only transmits data just once. 
So in order to transmit data continuously, I added Radio.Send(Buffer, BufferSize); in line 111 in the static void OnTxDone(void) function. Do you think this is the right way to send data continuously? Or should I modify it?