STMicroelectronics/STM32CubeWL #5: "disable" macro calls "enable" function

wolsty7, 2021-02-19:
It doesn't appear to be used anywhere but this bug is in each board_resources.h file accross the application examples.
https://github.com/STMicroelectronics/STM32CubeWL/blob/747d4e27df07dcfc995b54f8af8a0bee38dfef8d/Projects/NUCLEO-WL55JC/Applications/LoRaWAN/LoRaWAN_End_Node/Core/Inc/board_resources.h#L97
https://github.com/STMicroelectronics/STM32CubeWL/blob/747d4e27df07dcfc995b54f8af8a0bee38dfef8d/Projects/NUCLEO-WL55JC/Applications/LoRaWAN/LoRaWAN_End_Node/Core/Inc/board_resources.h#L115
https://github.com/STMicroelectronics/STM32CubeWL/blob/747d4e27df07dcfc995b54f8af8a0bee38dfef8d/Projects/NUCLEO-WL55JC/Applications/LoRaWAN/LoRaWAN_End_Node/Core/Inc/board_resources.h#L133

ASELSTM, 2021-02-22:
Hi @wolsty7,

The issue you pointed out has been confirmed, a fix will be implemented and made available in future release. Thank you once again for your contribution.

With regards,

ASELSTM, 2021-02-22:
ST Internal Reference: 101828

ASELSTM, 2021-08-16:
Hi @wolsty7,

The issue you reported has been fixed in the frame of version v1.1.0 of the STM32CubeWL published on GitHub.

Please allow me then to close this thread and thank you again for having reported.

With regards,

closed 2021-08-16 by ASELSTM