STMicroelectronics/STM32CubeWL #7: LoRaWAN end node issue in dual core and single core project with CubeIDE

CandraSN, 2021-04-11:
Hi
I'm using Nucleo-WL55JC1 STM32WL development board for LoRaWAN end node dual core and i have a board with MCU STM32WLE5CB for LoRaWAN end node single core.

I have configured LoRaWAN end node protocol parameters like this documentation (https://www.st.com/resource/en/application_note/dm00660451-how-to-build-a-lora-application-with-stm32cubewl-stmicroelectronics.pdf). 

The program from the example project has been uploaded successfully. After I uploaded the LoRaWAN end node program to the STM32WL55JC1 (dual core) development board, it said "warning: difference between the mapping table placement in memory bla...bla..." on the external terminal via serial usb and couldn't send data to the server / gateway with OTAA.

Meanwhile, for the single core STM32WLECB MCU end node, the program was successfully uploaded with the Lora protocol configured. But there is no response / action from the node sending data to the server / gateway with OTAA. On this board, I'm not using TCXO and DCDC.

So, from the problem I got, is there a solution or configuration that I missed so that the nodes can send data with OTTA to the server / gateway. I really appreciate your attention, thank you.

ASELSTM, 2021-04-19:
Hi @CandraSN,

Thank you for you contribution.

Please allow me to introduce @YoannBo who will comment your post and provide you with more details about this point.

With regards,

CandraSN, 2021-04-19:
> Hi @CandraSN,
> 
> Thank you for you contribution.
> 
> Please allow me to introduce @YoannBo who will comment your post and provide you with more details about this point.
> 
> With regards,

Hi @ASELSTM 

Maybe i want to add something
So, I have tried using Firmware Package v.0.4.0 + EWARM (IAR) toolchain and the result is that I can transmit data in OTAA and ABP modes with end node single core program. But, by using Firmware Package v.1.0.0 + cubeIDE toolchain with single core mcu, I have not succeeded in using it for data transmission to the gateway. I've configured it based on the following documentation (https://www.st.com/resource/en/application_note/dm00660451-how-to-build-a-lora-application-with-stm32cubewl-stmicroelectronics.pdf). After I uploaded the single core end node program to my custom board (STM32WLE5CBU6), there was no response from the board or gateway.

I focus on development with a single core STM32WL with this firmware package.

I'm really looking forward for your reply, thank you.

ASELSTM, 2021-04-21:
Hi,

@YoannBo thank you for your reply and for the provided details.

@CandraSN since the same issue has already been reported and answered in the **ST Community**, please allow me then to close this thread. Thank you for you understanding.

With regards,

closed 2021-04-21 by ASELSTM