STMicroelectronics/STM32CubeWL #8: SBSFU_2_Images_DualCore  CM0PLUS compiler error

fatihyasar, 2021-04-27:
Hello there,
I am trying to compile the SBSFU_2_Images_DualCore project.
Although I did it in the correct order according to the document "Getting started with the SBSFU of STM32CubeWL (UM2767)". I am getting a compiler error.

I'm sure that, compiling the correct order according to Getting started with the SBSFU of STM32CubeWL (UM2767)

Compile order is;

2_Images_SECoreBin - compile success
2_Images_KMS_Blob - compile success
2_Images_SBSFU
2_Images_SBSFU_CM0PLUS - compile success
2_Images_SBSFU_CM4) - compile success
2_Images_UserApp_CM4 - compile success
2_Images_UserApp_CM0PLUS - compiler error

The error is multiple definition error .

 

c:\st\stm32cubeide_1.6.0\stm32cubeide\plugins\com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.9-2020-q2-update.win32_1.5.0.202011040924\tools\arm-none-eabi\bin\ld.exe: ../../../2_Images_SBSFU/STM32CubeIDE/CM0PLUS/Debug\se_interface_app.o: in function `SE_KMS_CloseSession':

(.SE_IF_Code+0xbd8): multiple definition of `SE_KMS_CloseSession'; ../../../2_Images_SBSFU/STM32CubeIDE/CM0PLUS/Debug\se_interface_app.o:(.SE_IF_Code+0xbd8): first defined here

c:\st\stm32cubeide_1.6.0\stm32cubeide\plugins\com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.9-2020-q2-update.win32_1.5.0.202011040924\tools\arm-none-eabi\bin\ld.exe: ../../../2_Images_SBSFU/STM32CubeIDE/CM0PLUS/Debug\se_interface_app.o: in function `SE_KMS_DigestInit':

(.SE_IF_Code+0x121c): multiple definition of `SE_KMS_DigestInit'; ../../../2_Images_SBSFU/STM32CubeIDE/CM0PLUS/Debug\se_interface_app.o:(.SE_IF_Code+0x121c): first defined here

c:\st\stm32cubeide_1.6.0\stm32cubeide\plugins\com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.9-2020-q2-update.win32_1.5.0.202011040924\tools\arm-none-eabi\bin\ld.exe: ../../../2_Images_SBSFU/STM32CubeIDE/CM0PLUS/Debug\se_interface_app.o: in function `SE_KMS_EncryptInit':

(.SE_IF_Code+0xedc): multiple definition of `SE_KMS_EncryptInit'; ../../../2_Images_SBSFU/STM32CubeIDE/CM0PLUS/Debug\se_interface_app.o:(.SE_IF_Code+0xedc): first defined here

c:\st\stm32cubeide_1.6.0\stm32cubeide\plugins\com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.9-2020-q2-update.win32_1.5.0.202011040924\tools\arm-none-eabi\bin\ld.exe: ../../../2_Images_SBSFU/STM32CubeIDE/CM0PLUS/Debug\se_interface_app.o: in function `SE_KMS_DestroyObject':

(.SE_IF_Code+0xc94): multiple definition of `SE_KMS_DestroyObject'; ../../../2_Images_SBSFU/STM32CubeIDE/CM0PLUS/Debug\se_interface_app.o:(.SE_IF_Code+0xc94): first defined here

c:\st\stm32cubeide_1.6.0\stm32cubeide\plugins\com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.9-2020-q2-update.win32_1.5.0.202011040924\tools\arm-none-eabi\bin\ld.exe: ../../../2_Images_SBSFU/STM32CubeIDE/CM0PLUS/Debug\se_interface_app.o: in function `SE_KMS_Encrypt':

(.SE_IF_Code+0xf3c): multiple definition of `SE_KMS_Encrypt'; ../../../2_Images_SBSFU/STM32CubeIDE/CM0PLUS/Debug\se_interface_app.o:(.SE_IF_Code+0xf3c): first defined here

 
I didn't do any code change. I just downloaded and opened in STM32CubeIde.

Is this related with STM32Cube_FW_WL_V1.0.0 ?

 What's wrong ?

fatihyasar, 2021-04-27:
Thank you for accepting as issue. 

Is there any workaround to fix this problem?

RKOUSTM, 2021-05-07:
Hi @fatihyasar,

Thank you for this report. This problem has been detected in later versions of Cube IDE. For version `v1.50`, no problem has been detected.

Actually, the point you raised has already been dealt and fixed internally. The fix will be made available in the frame of a future release `STM32CubeWL v1.1.0`.

Thank you once more for your contribution.

With regards

RKOUSTM, 2021-05-07:
ST Internal Reference: 100823

RKOUSTM, 2021-08-16:
Hi @fatihyasar,

The fix you requested has been implemented and is now available in the frame of the latest STM32CubeWL package V1.1.0 release. 

This issue can be closed now. Thank you again for your contribution.

With regards,

closed 2021-08-16 by RKOUSTM