STMicroelectronics/STM32CubeWL #9: `.text' will not fit in region `ROM'

fatihyasar, 2021-04-27:
if I'm trying to compile the LoRaWAN_End_Node_DualCore project. I compiled the LoRaWAN_End_Node_DualCore_CM4 project. However, I cannot compile the LoRaWAN_End_Node_DualCore_CM0PLUS project.

Compiler error is; 
/Applications/STM32CubeIDE.app/Contents/Eclipse/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.9-2020-q2-update.macos64_1.5.0.202011040924/tools/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: **LoRaWAN_End_Node_DualCore_CM0PLUS.elf section `.text' will not fit in region `ROM'**
/Applications/STM32CubeIDE.app/Contents/Eclipse/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.9-2020-q2-update.macos64_1.5.0.202011040924/tools/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: region `ROM' overflowed by 7240 bytes
collect2: error: ld returned 1 exit status
make: *** [makefile:76: LoRaWAN_End_Node_DualCore_CM0PLUS.elf] Error 1
"make -j7 all" terminated with exit code 2. Build might be incomplete.

fatihyasar, 2021-04-27:
Same compiler problem on LoRaWAN_AT_Slave_DualCore. It happening only on M0+

I tested both STM32CubeIde installed Windows and MacOS Catalina.

ASELSTM, 2021-04-29:
Hi @fatihyasar,

Thank you for you contribution.

Please allow me to introduce @YoannBo who will comment your post and provide you with more details about this point.

With regards,

YoannBo, 2021-05-03:
Hi,

please try to compile with optimization for size on CM0+

ASELSTM, 2021-05-18:
Hi @fatihyasar,

Any update regarding this issue ?  Were you able to fix compilation errors ? 

With regards,

ASELSTM, 2021-06-09:
Hi @fatihyasar,

Please allow me to close this thread as no activity has been registered since a month. You can reopen it at any time if you have details to provide us to help you.

Thank you for your comprehension and thank you once more for your contribution

With regards,

closed 2021-06-09 by ASELSTM