STMicroelectronics/STM32CubeWL #10: Custom sleep and wakeup system on the lora end node STM32WL

CandraSN, 2021-05-02:
Hi,
I am a bit confused on the lora STM32WL sequence protocol system, I want my device to sleep and wakeup according to the periodic values I want. I have made my own sleep function (stop 2 mode & radio sleep) with wake up triggered using RTC (in seconds) and push button. Where can I put my function into the program lora end node STM32WL ?

Thank you

ASELSTM, 2021-05-10:
Thank you for you contribution.

Please allow me to introduce @YoannBo who will comment your post and provide you with more details about this topic.

With regards,

YoannBo, 2021-05-10:
Hello,

When the system has nothing to do,  the system calls the  idle task: UTIL_SEQ_Idle is located into sys_app.c.
In the CubeWL package, the `UTIL_SEQ_Idle` calls directly `UTIL_LPM_EnterLowPower`

Functions in `PWR_EnterStopMode`, `PWR_ExitStopMode `and equivalent SleepMode (`PWR_EnterSleepMode`, `PWR_ExitSleepMode`) are defined in  `stm32_lpm_if.c`. They can be re-written depending on the application.

Now in our system, this is already managed. If you want to periodically wake-up, you can create a timer, for instance toggling a LED

```
/**
  * @brief  User timer callback function
  * @param  context
  * @retval none
  */
static void OnUserTimerEvent(void *context);

/**
  * @brief Timer to handle the application Join Led to toggle
  */
static UTIL_TIMER_Object_t UserTimer;

void yourFunction(void)
{
  /*Create a UserTimer toggling every 5 seconds*/
  UTIL_TIMER_Create(&UserTimer, 5000, UTIL_TIMER_PERIODIC, OnUserTimerEvent, NULL);
  /* Starts UserTimer*/
  UTIL_TIMER_Start(&UserTimer);
}
/* OnUserTimerEvent */
static void OnUserTimerEvent(void *context)
{
  BSP_LED_Toggle(LED_RED) ;
}
```
Please be aware, that you should not call APP_PRINTF before entering to stop mode. Indeed, any time everything is printed out, it will wake up the system. and enter again. This will certificate an infinite loop.

Best Regards

YoannBo, 2021-05-12:
Hello @CandraSN ,

from what I read, this is already done in the LoRaWAN_End_Node project :
Join -> send data frame -> stop2-> wake-up -> send data frame -> stop2
by default, the  LoRaWAN_End_Node project sends a Tx frame every 10 seconds (when Duty Cycle allows)
Duty cycles can be updated in lora_app.h
#define APP_TX_DUTYCYCLE                            10000

Also, by default, a timer triggers the send_request. But it can also be triggered by nucleo button sw1 by change line 195 in lora_app.c
/* send will be done if on timer*/
static TxEventType_t EventType = TX_ON_TIMER;
to
/* send will be done if you push sw1 */
static TxEventType_t EventType = TX_ON_EVENT;

let me know

YoannBo, 2021-05-12:
additionnal note:

The RTC is used (in binary mode new!) directly by the timer (stm32_timer.c) used by the stack
I would advise not to override it.

CandraSN, 2021-05-17:
Hi @YoannBo 

After I follow your instructions and did some little experiments, I understand how to modify the code based on my system. Thank you very much, your informations are very clear.

Best Regards

CandraSN, 2021-05-20:
btw, sorry out of topic
how can I print float/double data with APP_PRINTF ?

YoannBo, 2021-05-20:
Hi @CandraSN 

in file _utilities_conf.h_ line
`#define UTIL_ADV_TRACE_VSNPRINTF(...)              tiny_vsnprintf_like(__VA_ARGS__)   `
Can you change `tiny_vsnprintf_like` to `vsnprintf` (bigger in size) from _stdio.h_ ?

I believe it should be ok,

Best Regards

CandraSN, 2021-05-24:
Hi @YoannBo 
Yes, problem solved using sprintf and transmit to serial with APP_PRINTF

Thank you

ASELSTM, 2021-05-24:
Hi,

@YoannBo Thank you for these clarifications.

@CandraSN Since the issue is answered and the problem is solved, please allow me to close this thread. Thank you once more for your contribution.

With regards,

closed 2021-05-24 by ASELSTM

tcpipchip, 2023-09-30:
thanks to this info. i can use in on st lorawan SDK

TiagoFernandes271089, 2023-10-03:
Hello,

Thanks to friends for sharing the information about the LoRaWAN_End_Node code. I am also using it to develop a project here at the company.
I would like to ask some questions please...
1st Question: I am using this example to wake up the module in loop mode. I would like to exit the loop module by pressing a button and keep the module constantly awake. Could you help me with an example, please.
2nd Question: Being in loop mode, I would like to transmit (uplink) data to the network server each time the module wakes up. Could you also help me with an example, please.

Thank you!

tcpipchip, 2023-10-03:
Tiago, your problem now is that 40uA consumption, right ?

TiagoFernandes271089, 2023-10-03:
Yes.
In addition to these 2 questions above, I also have this problem with consumption remaining around 40uA.

e3dz, 2023-11-28:
Hello @TiagoFernandes271089 and @tcpipchip , i have the same issue with stop2mode and consumption at around 40uA.

Did you manage to solve the current consumption problem?

My application is based on a Lora-E5 and i have verified with the factory firmware that it can run on approx. 3uA(@3.6V). When using following code below i reach a 40uA(@3.6V) current consumption.
```c
#include "main.h"

#include "stm32wlxx_hal_pwr_ex.h"

void SystemClock_Config(void);
static void MX_GPIO_Init(void);

int main(void)
{
  HAL_Init();

  SystemClock_Config();

  MX_GPIO_Init();

  HAL_PWREx_EnterSTOP2Mode(PWR_STOPENTRY_WFI);

  while (1)
  {
  }
}

void SystemClock_Config(void)
{
  RCC_OscInitTypeDef RCC_OscInitStruct = {0};
  RCC_ClkInitTypeDef RCC_ClkInitStruct = {0};

  HAL_PWR_EnableBkUpAccess();
  __HAL_RCC_LSEDRIVE_CONFIG(RCC_LSEDRIVE_LOW);

  __HAL_PWR_VOLTAGESCALING_CONFIG(PWR_REGULATOR_VOLTAGE_SCALE2);

  RCC_OscInitStruct.OscillatorType = RCC_OSCILLATORTYPE_LSE|RCC_OSCILLATORTYPE_MSI;
  RCC_OscInitStruct.LSEState = RCC_LSE_ON;
  RCC_OscInitStruct.MSIState = RCC_MSI_ON;
  RCC_OscInitStruct.MSICalibrationValue = RCC_MSICALIBRATION_DEFAULT;
  RCC_OscInitStruct.MSIClockRange = RCC_MSIRANGE_6;
  RCC_OscInitStruct.PLL.PLLState = RCC_PLL_NONE;
  if (HAL_RCC_OscConfig(&RCC_OscInitStruct) != HAL_OK)
  {
    Error_Handler();
  }

  RCC_ClkInitStruct.ClockType = RCC_CLOCKTYPE_HCLK3|RCC_CLOCKTYPE_HCLK
                              |RCC_CLOCKTYPE_SYSCLK|RCC_CLOCKTYPE_PCLK1
                              |RCC_CLOCKTYPE_PCLK2;
  RCC_ClkInitStruct.SYSCLKSource = RCC_SYSCLKSOURCE_MSI;
  RCC_ClkInitStruct.AHBCLKDivider = RCC_SYSCLK_DIV1;
  RCC_ClkInitStruct.APB1CLKDivider = RCC_HCLK_DIV1;
  RCC_ClkInitStruct.APB2CLKDivider = RCC_HCLK_DIV1;
  RCC_ClkInitStruct.AHBCLK3Divider = RCC_SYSCLK_DIV1;

  if (HAL_RCC_ClockConfig(&RCC_ClkInitStruct, FLASH_LATENCY_0) != HAL_OK)
  {
    Error_Handler();
  }
}

static void MX_GPIO_Init(void)
{
  GPIO_InitTypeDef GPIO_InitStruct = {0};

  __HAL_RCC_GPIOA_CLK_ENABLE();
  __HAL_RCC_GPIOB_CLK_ENABLE();
  __HAL_RCC_GPIOC_CLK_ENABLE();
  __HAL_RCC_GPIOH_CLK_ENABLE();

  GPIO_InitStruct.Pin = GPIO_PIN_All;
  GPIO_InitStruct.Mode = GPIO_MODE_ANALOG;
  GPIO_InitStruct.Pull = GPIO_NOPULL;

  HAL_GPIO_Init(GPIOA, &GPIO_InitStruct);
  HAL_GPIO_Init(GPIOB, &GPIO_InitStruct);
  HAL_GPIO_Init(GPIOC, &GPIO_InitStruct);
  HAL_GPIO_Init(GPIOH, &GPIO_InitStruct);

  __HAL_RCC_GPIOA_CLK_DISABLE();
  __HAL_RCC_GPIOB_CLK_DISABLE();
  __HAL_RCC_GPIOC_CLK_DISABLE();
  __HAL_RCC_GPIOH_CLK_DISABLE();
}
```

TiagoFernandes271089, 2023-11-28:
Hello @e3dz ,

I managed to resolve it here.
In my case, whenever I pressed the button (PA12), the module woke up and never returned to sleep mode (+/-5uA).
I was calling the "void EXTI1_IRQHandler(void)" interrupt function.
I modified my function to "void EXTI15_10_IRQHandler(void)"

/* EXTI interrupt initialization */
HAL_NVIC_SetPriority(EXTI0_IRQn, 0, 0);
HAL_NVIC_EnableIRQ(EXTI0_IRQn);
HAL_NVIC_SetPriority(EXTI1_IRQn, 0, 0);
HAL_NVIC_EnableIRQ(EXTI1_IRQn);
HAL_NVIC_SetPriority(EXTI9_5_IRQn, 0, 0);
HAL_NVIC_EnableIRQ(EXTI9_5_IRQn);
HAL_NVIC_SetPriority(EXTI15_10_IRQn, 0, 0);
HAL_NVIC_EnableIRQ(EXTI15_10_IRQn);

Also check if any portal is not connected in sleep mode.
In my case, I had to call the function to disable UART2:
HAL_UART_DeInit(&huart2);

tcpipchip, 2023-11-28:
haha...the answered when i was typing :)

TiagoFernandes271089, 2023-11-28:
Hello,

Follows as requested:

#include "main.h"
#include "i2c.h"
#include "app_lorawan.h"
#include "tim.h"
#include "gpio.h"
#include "sys_app.h"
#include "app_version.h"
#include "stm32_lpm.h"		//Include para a função "UTIL_LPM_EnterLowPower()" (Tiago)
#include "utilities_conf.h"
#include "stm32_lpm_if.h"	//Include para a Função "PWR_EnterSleepMode()" (Tiago)
#include "lora_app.h"

/* Private includes ----------------------------------------------------------*/
/* USER CODE BEGIN Includes */
static void OnUserTimerEvent(void *context);
static UTIL_TIMER_Object_t UserTimer;

/* USER CODE END Includes */
extern void HAL_GPIO_EXTI_Callback(uint16_t GPIO_Pin);

extern UART_HandleTypeDef huart2;

void SystemClock_Config(void);

int main(void)
{
	/* Reset of all peripherals, Initializes the Flash interface and the Systick. */
	HAL_Init();
	/* Configure the system clock */
	SystemClock_Config();
	/* Initialize all configured peripherals */
	//MX_LoRaWAN_Init();
	SystemApp_Init();		//Tiago: Aqui inicia o Hall
	MX_GPIO_Init();
	MX_I2C1_Init();
	MX_TIM16_Init();
	HAL_TIM_Base_Start_IT(&htim16);

	/* Infinite loop */
	/* USER CODE BEGIN WHILE */

	HAL_UART_DeInit(&huart2);

	while (1)
	{
            MX_LoRaWAN_Process();
	}

}

void SystemClock_Config(void)
{
	RCC_OscInitTypeDef RCC_OscInitStruct = {0};
	RCC_ClkInitTypeDef RCC_ClkInitStruct = {0};

	/** Configure LSE Drive Capability
	 */
	HAL_PWR_EnableBkUpAccess();
	__HAL_RCC_LSEDRIVE_CONFIG(RCC_LSEDRIVE_LOW);

	/** Configure the main internal regulator output voltage
	 */
	__HAL_PWR_VOLTAGESCALING_CONFIG(PWR_REGULATOR_VOLTAGE_SCALE1);

	/** Initializes the CPU, AHB and APB buses clocks
	 */
	RCC_OscInitStruct.OscillatorType = RCC_OSCILLATORTYPE_LSE|RCC_OSCILLATORTYPE_MSI;
	RCC_OscInitStruct.LSEState = RCC_LSE_ON;
	RCC_OscInitStruct.MSIState = RCC_MSI_ON;
	RCC_OscInitStruct.MSICalibrationValue = RCC_MSICALIBRATION_DEFAULT;
	RCC_OscInitStruct.MSIClockRange = RCC_MSIRANGE_11;//RCC_MSIRANGE_8
	RCC_OscInitStruct.PLL.PLLState = RCC_PLL_NONE;
	if (HAL_RCC_OscConfig(&RCC_OscInitStruct) != HAL_OK)
	{
		Error_Handler();
	}

	/** Configure the SYSCLKSource, HCLK, PCLK1 and PCLK2 clocks dividers
	 */
	RCC_ClkInitStruct.ClockType = RCC_CLOCKTYPE_HCLK3|RCC_CLOCKTYPE_HCLK
			|RCC_CLOCKTYPE_SYSCLK|RCC_CLOCKTYPE_PCLK1
			|RCC_CLOCKTYPE_PCLK2;
	RCC_ClkInitStruct.SYSCLKSource = RCC_SYSCLKSOURCE_MSI;
	// RCC_ClkInitStruct.AHBCLKDivider = RCC_SYSCLK_DIV1;
	RCC_ClkInitStruct.AHBCLKDivider = RCC_SYSCLK_DIV10;//DIV10 = 4,8MHz
	//RCC_ClkInitStruct.APB1CLKDivider = RCC_HCLK_DIV1;
	RCC_ClkInitStruct.APB1CLKDivider = RCC_HCLK_DIV16;
	RCC_ClkInitStruct.APB2CLKDivider = RCC_HCLK_DIV1;
	RCC_ClkInitStruct.AHBCLK3Divider = RCC_SYSCLK_DIV1;

	if (HAL_RCC_ClockConfig(&RCC_ClkInitStruct, FLASH_LATENCY_2) != HAL_OK)
	{
		Error_Handler();
	}
}

void HAL_TIM_PeriodElapsedCallback(TIM_HandleTypeDef *htim)
{
	if(htim == &htim16)
	{
		//HAL_GPIO_TogglePin(DEF_LED_BLUE_GPIO_Port, DEF_LED_BLUE);
		if(St_Timer.t_200ms)St_Timer.t_200ms--;
		if(St_Timer.t_1s)St_Timer.t_1s--;
		if(St_Timer.t_3s)St_Timer.t_3s--;
		if(St_Timer.t_5s)St_Timer.t_5s--;
		if(St_Timer.t_6s)St_Timer.t_6s--;
		if(St_Timer.t_10s)St_Timer.t_10s--;
		if(St_Timer.t_out)St_Timer.t_out--;

	}
}

/* USER CODE END 4 */

/**
 * @brief  This function is executed in case of error occurrence.
 * @retval None
 */
void Error_Handler(void)
{
	/* USER CODE BEGIN Error_Handler_Debug */
	/* User can add his own implementation to report the HAL error return state */
	__disable_irq();
	while (1)
	{
	}
	/* USER CODE END Error_Handler_Debug */
}

#ifdef  USE_FULL_ASSERT
/**
 * @brief  Reports the name of the source file and the source line number
 *         where the assert_param error has occurred.
 * @param  file: pointer to the source file name
 * @param  line: assert_param error line source number
 * @retval None
 */
void assert_failed(uint8_t *file, uint32_t line)
{
	/* USER CODE BEGIN 6 */
	/* User can add his own implementation to report the file name and line number,
     ex: printf("Wrong parameters value: file %s on line %d\r\n", file, line) */
	while (1)
	{
	}
	/* USER CODE END 6 */
}
#endif /* USE_FULL_ASSERT */