STMicroelectronics/STM32CubeWL #11: stm32wle stuck during software reset

PG430, 2021-05-25:
Hi,

I'm using STM32WLE5JCI6 based LORA module.  I need to used the software reset. using v1.0.0 release.
I tried the “HAL_NVIC_SystemReset()” provided with source, but most of the times it doesn’t work,

- 1 out of 3 times, the reset is working and the board restart correctly,
- 2 out of 3 times, the call to “HAL_NVIC_SystemReset()” freezing the microcontroller. Nothing can be done any more except hardware reset.

Is this a known problem? is there any solution for this problem?

PG430, 2021-05-31:
Thank you for accepting this problem,

Is there any fix for this issue?

PG430, 2021-06-02:
@ASELSTM any update on this?

PG430, 2021-06-10:
@ASELSTM  is it a known issue? is there any way to solve this issue?

ASELSTM, 2021-06-10:
Hi @PG430,

The `HAL_NVIC_SystemReset()` is actually a complete reuse of ARM CMSIS API `NVIC_SystemReset()`, there should be no issue with the use of this API. The NVIC reset is not only performing a CPU reset but also induces a complete system one. Your issue could be related to a bad boot configuration. 
For a better understanding of your issue may I ask you few questions : 

1. How do you consider the target locked up ? 

2. Would you verify the boot configuration (option bytes/pins) to make sure you don't boot on Flash or SRAM ? 
Please make sure that IDE is not used for that as some configuration disturbs normal scenario.

3. When done, could you add a reset reason vector catch at startup ? Entering a basic while loop if RCC_CSR_SFTRSTF  is different from 0, the while loop could be exit by the exit button. Please make sure you clear all reset flags after reading them. 

With regards,

PG430, 2021-06-10:
Hi @ASELSTM 

1. How do you consider the target locked up ?
[image]

2.Would you verify the boot configuration (option bytes/pins) to make sure you don't boot on Flash or SRAM ?
Please make sure that IDE is not used for that as some configuration disturbs normal scenario.

[image]

I've not change anything in Option Bytes, Those settings are same as provided in GIT Release v1.0.0

3. When done, could you add a reset reason vector catch at startup ? Entering a basic while loop if RCC_CSR_SFTRSTF is different from 0, the while loop could be exit by the exit button. Please make sure you clear all reset flags after reading them.
  **[PG]**: I'll update soon

Regards,

ASELSTM, 2021-09-03:
Hi @PG430,

Would you please elaborate on what you were faced with when you have said that the “MCU is frozen”.

With regards,