STMicroelectronics/STM32CubeWL #12: the BUSY switching time so long to 10ms

wangpengli, 2021-05-31:
After radio.init called, we called the radio.rx(0xffffff) for into the rx mode , but when I test it ,I found the switching time is 10ms. 
 
the code like this:

tick1 = HAL_Gettick();
Radio.Rx(0xFFFFFF);
tick2 - HAL_Gettick();
printf(“switch time = %d\n”,tick2-tick1);

[image]

wangpengli, 2021-05-31:
I check the busy switching time in the RM0453 document, and checked it from SX126X DS , found like blow:

[image]

the switching time should more less than the 1ms. 

Pls check it , thank you.

wangpengli, 2021-05-31:
I use the pingpong demo app and the JC1 devkit board for test. and if I add set-standby use the STDBY_XOSC instead of the default STDBY_RC, the issue is same. 

I also try use the SetRxTxFallbakMode set to XOSC , the issue is same too.

wangpengli, 2021-06-01:
I made the same test case use STM32MCU + SX1268, no problem ,all of  the switching time less than 1ms.

YoannBo, 2021-06-10:
Hi, 

on the nucleo, there is a TCXO. The TCXO requires a wake up time. 
in the cubeWL_1.0.0, the TCXO_WAKE_UP_TIME was set at 10ms, Probably this is what you see.

To change this, you can go in SubGHz_Phy/target/radio_board_if.h
/* Radio maximum wakeup time (in ms) */
#define RF_WAKEUP_TIME                     10U

This 10 can be lowered to 1ms without impacting the RF performances.

wangpengli, 2021-06-15:
OK, Got it , thank you , I changed it and seems the issue is resolved .

RKOUSTM, 2021-06-22:
Hi @wangpengli,

Regarding the point for which you opened the issue, as it is resolved, please allow me to close the issue.

With regards,

closed 2021-06-22 by RKOUSTM