STMicroelectronics/STM32CubeWL #13: STM32WL: In QC_DualCore Firmware CM0PLUS binary size is too large

PG430, 2021-06-04:
Hello,

STM32Wl dual core ins using M0+ and M4. GIT release v1.0.0 has firmware for both.
but when we compiled the M0+ source binary size is unexpectedly too large 
Bin size: 393126Kb 
ELF size: 2588Kb
He size: 196Kb

[image]

Now to reduce the size of binary i've made changes in .ld file,

[image]

after .ld file change binary size is reduced in KB

[image]

PG430, 2021-06-04:
Thank you for accepting this problem,

Is there any fix for this issue?

ASELSTM, 2021-06-09:
Hi @PG430,

Thank you for you contribution.

Please allow me to introduce @YoannBo who will comment your post and provide you with more details about this point.

With regards,

YoannBo, 2021-06-10:
Hi @PG430,

You are right. This is implemented in the upcoming release (coming very soon),

thank you

PG430, 2021-06-11:
@YoannBo Thank you for the confirmation

closed 2021-06-11 by PG430