STMicroelectronics/STM32CubeWL #14: I am using STM Cube IDE for developement of the LoRa E5 development board which has the STM32wle5jc module in it. Since there is no example code for this board on the cube ide, i wanted to adapt the Subghz Ping Pong code for my board's pinout.

ashutosh952, 2021-06-09:
**Caution**
The Issues are strictly limited for the reporting of problem encountered with the software provided in this project.
For any other problem related to the STM32 product, the performance, the hardware characteristics and boards, the tools, or the environment in general, please post a topic in the [ST Community/STM32 MCUs forum](https://community.st.com/s/group/0F90X000000AXsASAW/stm32-mcus).

**Describe the set-up**
 * The board (either ST RPN reference or your custom board).
 * IDE or at least the compiler and its version.

**Describe the bug**
A clear and concise description of what the bug is.

**How To Reproduce**
1. Indicate the global behavior of your application project.

2. The modules that you suspect to be the cause of the problem (Driver, BSP, MW ...).

3. The use case that generates the problem.

4. How we can reproduce the problem.

**Additional context**
If you have a first analysis or patch correction, thank you to share your proposal.

**Screenshots**
If applicable, add screenshots to help explain your problem.

ashutosh952, 2021-06-09:
I am using STM Cube IDE for developement of the LoRa E5 development board which has the STM32wle5jc module in it. Since there is no example code for this board on the cube ide, i wanted to adapt the Subghz Ping Pong code for my board's pinout. The board gets disconnected from the debugger when we let LOW_POWER_DISABLE be 0 in sys_conf.h file. To debug it we have to set it to 1. Now when I ran the code step by step in the debug mode, I did not find the PingPong_Process being called repeatedly in the while(1){} loop of the code. This function is responsible to transmit and receive the ping or pong msg. Can someone help me on this?

ASELSTM, 2021-06-15:
Hi @ashutosh952,

Thank you for your contribution. Please allow me to introduce @YoannBo who will look deeper into your report.

Thank you for your patience and again thank you for your contribution.

With regards,

ashutosh952, 2021-06-16:
Sorry, I am not familiar with how to introduce someone to look into the issue, do I have to click on some option to do it? or it's just literal?

YoannBo, 2021-06-16:
Hi @ashutosh952 ,

I'm not sure I understand what problem you are facing.

in the while(1),  the MCU is in sleep mode (since you  LOW_POWER_DISABLE to 0 ) unless an Interrupt from the Radio is waking up MCU. 
The Radio IT (for instance OnRxDone, OnTxDone) are setting the CFG_SEQ_Task_PingPong_Process bitmap with `UTIL_SEQ_SetTask((1 << CFG_SEQ_Task_PingPong_Process), CFG_SEQ_Prio_0);.`
Before  the MCU enters sleep mode again, the PingPong_Process will be called.

Hope this helps,

Best Regards

ashutosh952, 2021-06-16:
Hi @YoannBo, thanks for the reply. Could you also explain in brief what is the use of `UTIL_SEQ_SetTask` function and how does it work?

YoannBo, 2021-06-16:
See below:

Sequencer is a 'package while (1) loop' which is race conditions safe:
[image]

Hope it clarifies

ashutosh952, 2021-06-17:
I tried to debug my code line by line, function by function using a stlink v2 clone in STM Cube IDE, and this sequencer way of calling the function repeatedly did not work as expected since it did not even enter the ping pong process function in the function`UTIL_SEQ_Run` in while (1) loop. But when I use it without debugging, then it actually behaves like the function is being called as it should be and comport output of the Ping Pong example displays the msgs properly. Why is this happening? Can I not see the Ping Pong process being executed in the debug mode?

YoannBo, 2021-08-31:
Hello @ashutosh952 ,

did you succeed to debug? do you need further support?

best regards

ASELSTM, 2022-01-17:
Hi @ashutosh952,

Please allow me to close this thread as there is no activity since several months. You can reopen it at any time if you have more details so that we can help you.

With regards,

closed 2022-01-17 by ASELSTM