STMicroelectronics/STM32CubeWL #16: app data not sent notification in LmHandlerSend function (LmHandler lib)

krkoc, 2021-07-08:
I’ve run into a possible issue with the LmHandler library. STM32Cube_FW_WL_V1.0.0\Middlewares\Third_Party\LoRaWAN\LmHandler

In particular the LmHandlerSend function.

As I follow the code I see that before sending a frame a function calculates if the number of bytes to be sent will fit in a frame. It takes also in account the FOPTS field.

In case FOPTS field is empty the app data buffer of valid lenght gets sent and life is great.

In case the app data (valid lenght by itself) + FOPTS are greater than the total number of bytes allowed for the given region and spreading factor it will send an empty frame to “to flush MAC commands ”, as commented in the code.

What strikes me as problematic is not only that the app data buffer will be discarted but also that the user will not be informed that his/her buffer wasn’t sent.  LmHandlerSend will in this case return LORAMAC_HANDLER_SUCCESS

ASELSTM, 2021-07-12:
Hi @krkoc,

Please allow me to introduce @YoannBo who will look deeper into your report.

With regards,

krkoc, 2021-07-13:
Thanks. 

I could of course modify the library to get the information. I'm also curious to know if this is a real issue or is this somehow wanted behaviour.

ASELSTM, 2021-09-01:
ST Internal Reference: 112594

closed 2022-12-08 by ALABSTM