STMicroelectronics/STM32CubeWL #18: Task sequencer (stm32_seq) - priority and task number

RomanJasmann, 2021-08-05:
In utilities_def.h it is possible to add additional task priority numbers and task numbers:

```
/**
  * This is the list of priority required by the application
  * Each Id shall be in the range 0..31
  */
typedef enum
{
  CFG_SEQ_Prio_0,
  /* USER CODE BEGIN CFG_SEQ_Prio_Id_t */

  /* USER CODE END CFG_SEQ_Prio_Id_t */
  CFG_SEQ_Prio_NBR,
} CFG_SEQ_Prio_Id_t;

/**
  * This is the list of task id required by the application
  * Each Id shall be in the range 0..31
  */
typedef enum
{
  CFG_SEQ_Task_LmHandlerProcess,
  CFG_SEQ_Task_LoRaSendOnTxTimerOrButtonEvent,
  /* USER CODE BEGIN CFG_SEQ_Task_Id_t */

  /* USER CODE END CFG_SEQ_Task_Id_t */
  CFG_SEQ_Task_NBR
} CFG_SEQ_Task_Id_t;
```
However, the size of TaskPrio is set to a fixed value of 2:

```
/**
 * @brief default value of priority number.
 */
#ifndef UTIL_SEQ_CONF_PRIO_NBR 
  #define UTIL_SEQ_CONF_PRIO_NBR  (2)
#endif

/**
 * @brief task prio management.
 */
static UTIL_SEQ_Priority_t TaskPrio[UTIL_SEQ_CONF_PRIO_NBR];
```
This results in a hard fault when you add more than two additional task priority numbers without changing UTIL_SEQ_CONF_PRIO_NBR.

YoannBo, 2021-09-01:
Hello @RomanJasmann,

you are right, thanks for pointing this issue at us.
We should had somewhere in the utilities_conf.h:
/* USER CODE BEGIN EC */
#include "utilities_def.h"
#define UTIL_SEQ_CONF_PRIO_NBR CFG_SEQ_Prio_NBR
/* USER CODE END EC */

let me know if this works for you

best regards

RKOUSTM, 2021-09-02:
Hi @RomanJasmann,

Thank you for your reports. An internal tracker has been logged and a fix will be implemented and made available in the frame of a future release to modify the priority and the task number in the utilities_conf.h file according to your application as below:

> #define UTIL_SEQ_CONF_TASK_NBR                  (32)
> #define UTIL_SEQ_CONF_PRIO_NBR                  (2)

Also, you can refer to [this](https://github.com/STMicroelectronics/STM32CubeWB/blob/master/Projects/NUCLEO-WB15CC/Applications/BLE/BLE_p2pServer/Core/Inc/utilities_conf.h#L58-L59)  configuration file for STM32 utilities on the STM32CubeWB for more details.

Thank you again for you reports.

With regards,

RKOUSTM, 2021-09-02:
Hi @YoannBo,

Thank you for your answer and your contribution.

With regards,

RKOUSTM, 2021-09-02:
ST Internal Reference: 112597

closed 2022-12-08 by ALABSTM