STMicroelectronics/STM32CubeWL #20: DBGMCU_CR_TRACE_* defines missing from stm32wl*.h

aurel32, 2021-09-08:
**Describe the set-up**

 * Seeed Studio LoRa-E5 chip
 * gcc version 10.3.0

**Describe the bug**
The defines for the DBGMCU_CR register are missing for the TRACE_IOEN and TRACE_MODE bits. This is actually consistent with the reference manual RM0461 Rev 4 page 1284 where those bits also do not appears. However the reference manual definitely talk about the TRACE_SWO functionality in other places of the manual, and especially in section 36.

Now taking the bit definitions for this register from the STM32L4 (from which I am migrated my design), I can get the SWO functionality working perfectly. It therefore seems that this is only a documentation and header issue.

ASELSTM, 2021-09-21:
ST Internal Reference: 113930

aurel32, 2022-07-12:
Thanks for your answer. I agree that the STM32WL SoC does not have an ETM, but it has an ITM (which is the one I want to use), connected to the TPIU, which should be able to output the trace on the TRACESWO pin as described on figures 370 and 375 of RM0461. How to configure the TPIU to enable the TRACESWO pin?

adrienbruant, 2024-04-15:
I cannot get SWO to work unless I configure DBGMCU_CR as I would on a STM32L4, as per @aurel32. Please reconsider closing this ticket or provide a working sample logging with SWO.

aurel32, 2024-04-24:
I am using Zephyr, and this is the patch I have been using for quite sometimes. It should applies fine against Zephyr 3.6.0
https://gist.github.com/aurel32/fb946892873063b9adc535fdc6526124