STMicroelectronics/STM32CubeWL #21: LoRaWAN en node dual core issue receiving dowlink

Jeanbagarre, 2021-09-09:
Hello, I would like to report a bug using STM32CubeWL code for LoRaWAN end node dual core application. I'm using a NUCLEO-WL55JC1 with cube IDE.

I was trying to receive downlink message in order to toggle a LED. Radio communication did well but the payload was badly transmit to CM0 to CM4. Any message received given only the value 1.

After some research I corrected it by modifying LmHandler_mbwrapper.c line 708 

original code is :  next_addr = (appData->BufferSize + 7) & ~ 7;

I modify it as : next_addr **+**= (appData->BufferSize + 7);

Hope it could help someone with the same issue

YoannBo, 2021-09-16:
Hi @Jeanbagarre ,

we confirm it is a bug.
We must replace ‘=’ by ‘+=’ to prevent an override of third memcpy. But the ‘( xxx + 7 ) & ~7’ must be kept to force the 8-align (required by secure projects).

LoRaWAN_End_Node_DualCore\CM0PLUS\MbMux\LmHandler_mbwrapper.c

```
static void OnRxData_mbwrapper(LmHandlerAppData_t *appData, LmHandlerRxParams_t *params)
{
  /* USER CODE BEGIN OnRxData_mbwrapper_1 */

  /* USER CODE END OnRxData_mbwrapper_1 */
  MBMUX_ComParam_t *com_obj;
  uint32_t *com_buffer = NULL;
  uint8_t next_addr = 0;

  if (appData != NULL)
  {
    UTIL_MEM_cpy_8(aLoraMbWrapShare2Buffer, appData, sizeof(LmHandlerAppData_t));
    /* need to 8-align the next address */
    next_addr = (sizeof(LmHandlerAppData_t) + 7) & ~7;

    if (appData->BufferSize > 0)
    {
      UTIL_MEM_cpy_8(&aLoraMbWrapShare2Buffer[next_addr], appData->Buffer, appData->BufferSize);
      ((LmHandlerAppData_t *)aLoraMbWrapShare2Buffer)->Buffer = &aLoraMbWrapShare2Buffer[next_addr];
--      next_addr = (appData->BufferSize + 7) & ~7;
++      next_addr += (appData->BufferSize + 7) & ~7;
    }
  }

  if (params != NULL)
  {
    UTIL_MEM_cpy_8(&aLoraMbWrapShare2Buffer[next_addr], params, sizeof(LmHandlerRxParams_t));
  }
```

Jeanbagarre, 2021-09-17:
Hi Yoannbo,

thanks for your reply..

I've test your solution and I confirm bug is fixed.

thanks for help.

ASELSTM, 2021-09-20:
ST Internal Reference: 113752

closed 2022-12-08 by ALABSTM