STMicroelectronics/STM32CubeWL #27: Bash scripts missing from the repository

gregbreen, 2021-11-08:
I'm using **Nucleo-WL55JC1** and **STM32CubeIDE 1.6.0**. Building under **Ubuntu 20.04**.

Having cloned this repo, I successfully built and used the following two applications:

- https://github.com/STMicroelectronics/STM32CubeWL/tree/main/Projects/NUCLEO-WL55JC/Applications/LoRaWAN/LoRaWAN_AT_Slave/STM32CubeIDE
- https://github.com/STMicroelectronics/STM32CubeWL/tree/main/Projects/NUCLEO-WL55JC/Applications/LoRaWAN/LoRaWAN_End_Node/STM32CubeIDE

I'm now trying to use a FUOTA example, starting with **1_Image_SECoreBin**:

- https://github.com/STMicroelectronics/STM32CubeWL/tree/main/Projects/NUCLEO-WL55JC/Applications/LoRaWAN_FUOTA/1_Image_SECoreBin/STM32CubeIDE

However the build fails as follows:

```
/bin/sh: 1: ../prebuild.sh: not found
```

Searching the directory tree, I see that **prebuild.sh** is missing from all of the secure loading projects. Diving deeper, I see that the repository is missing **all** of the Bash **.sh** files. These files are present in the **e.stm32cubewl_v1-1-0_v1.1.0.zip** package that can be downloaded from the ST site. For example:

[image]

But absent from the repo:

[image]

gregbreen, 2021-11-08:
OK I see this Secure Engine utilities module is missing too, and explained here: https://github.com/STMicroelectronics/STM32CubeWL/blob/main/Middlewares/ST/STM32_Secure_Engine/readme.txt

So I guess the Bash scripts are omitted on purpose too.

RKOUSTM, 2021-11-15:
Hi @gregbreen,

Thank you for your contribution. You are absolutely right about this point. A fix will be made available as soon as possible.

Thank you once more for your contribution. 

With regards,

RKOUSTM, 2021-11-16:
Hi @gregbreen,

The fix is now available in GitHub. Please allow me then to close this issue now.

Thank you for your contribution. We are looking forward to reading from you again.

With regards,

closed 2021-11-16 by RKOUSTM