STMicroelectronics/STM32CubeWL #31: potential bug in SubGHz_Phy_PingPong example (LoRa Modulation)

ImanIDG, 2021-12-15:
Hi there.
I'm trying to run LoRa (NOT LoRaWAN) application. I even ran SubGHz_Phy_PingPong example code with no problem. I try to use my standard LoRaWAN gateway to follow up LoRa packets, but the thing is, my gateway couldn't recognize packets even though two stm32Nucleos can recognize each other's pockets.

If the SubGHz_Phy_PingPong example uses standard LoRa modulation(tell me if it doesn't), my gateway should recognize packets as proprietary LoRa packets. I think there is a bug on setting up LoRa modulation in SubGHz_Phy_PingPong example.

Do you have any suggestion for me or even same experience?

with best regards
iman

ASELSTM, 2022-01-25:
ST Internal Reference: 121509

ASELSTM, 2023-02-17:
Hi @ImanIDG,

The SubGHz_Phy_PingPong example is actually using the same lora modulation as the End_Node, a standard modulation. 

Concerning the gateway, as you said it is a LoRaWAN gateway that recognize LoRaWAN frames. 
If you send a payload with the LoRa modulation but without LoRaWAN encapsulation inside the payload, the gateway will receive it, but if there isn't the MAC headers needed, the payload is not recognized as LoRaWAN payload and will be sent to garbage.

It is understandable that the gateway would make a decision to keep or reject a packet. But the gateway will receive the LoRa packet and then transmit it to the network server that will process the payload to verify if it is a LoRa or a LoRaWAN packet(and to reject or keep it).

Moreover, in the case of the Ping Pong example, the frequency used as default in EU region to communicate between two end devices is 868.0 MHz, but a gateway at least listen on those 3 frequencies(in EU) : 868.1MHz , 868.3MHz and 868.5MHz. (cf RP002, section 2.4.2) so you will never see the packet if you keep the 868.0MHz frequency.

Maybe try to change the frequency used in the Ping Pong example(RF_FREQUENCY defined for 'REGION_EU868' in the file Application/SubGHz_Phy/App/subghz_phy_app.h), and try to check the gateway's log with the verbose at it maximum.

On our side it is working with the package STM32Cube_FW_WL and a gateway STM32F7 configure with The Things Network server. (don't forget to pass the AT command : 'AT+ECHO=ON' to view all the packets receive by the gateway)
Refer to UM2587 for the connection to the STM32F7 via USB, and the AT commands.

Hope this is helpful. Please allow me thus to close this thread.

With regards,

closed 2023-02-17 by ASELSTM