STMicroelectronics/STM32CubeWL #32: usart_if.c - overrun ISR flag has to be cleared before HAL_UART_Receive_IT

RomanJasmann, 2022-01-25:
After calling `vcom_Init`, the UART receiver is configured and ready to receive data. The problem is that when you later call `vcom_ReceiveInit`, the UART receiver might have already received some data and might also have raised an overrun error flag. So when you call `HAL_UART_Receive_IT` in `vcom_ReceiveInit`, the error interrupt enable flag is set and an error interrupt might be immediately triggered due to a possible unhandled overrun error. According to the HAL UART implementation, an overrun error is a blocking error, and the Rx interrupt handling will be stopped. Even if you implement the `HAL_UART_ErrorCallback ` and restart the Rx process in the error callback, it won't work because `HAL_UART_Receive_IT` in `vcom_ReceiveInit` will lock the UART handle resource by `__HAL_LOCK` and the `HAL_UART_Receive_IT` in the `HAL_UART_ErrorCallback` will encounter a locked handle resource.

```
void HAL_UART_ErrorCallback(UART_HandleTypeDef *huart) {
        if (huart->Instance==USART2) {
		if ((huart->ErrorCode & HAL_UART_ERROR_ORE) != 0) {
			HAL_UART_Receive_IT(&huart2, &charRx, 1);
		}
	}
}
```

FIX: At least clear the overrun ISR flag after polling the `USART_ISR_BUSY` in `vcom_ReceiveInit` and before starting the Rx interrupt handling by `HAL_UART_Receive_IT`.

```
UTIL_ADV_TRACE_Status_t vcom_ReceiveInit(void (*RxCb)(uint8_t *rxChar, uint16_t size, uint8_t error))
{

 ...

  /* Make sure that no UART transfer is on-going */
  while (__HAL_UART_GET_FLAG(&huart2, USART_ISR_BUSY) == SET);

  /* FIX */
  __HAL_UART_CLEAR_FLAG(&huart2, (UART_CLEAR_PEF | UART_CLEAR_FEF | UART_CLEAR_NEF | UART_CLEAR_OREF));
  __HAL_UART_SEND_REQ(&huart2, UART_RXDATA_FLUSH_REQUEST);

  /* Make sure that UART is ready to receive)   */
  while (__HAL_UART_GET_FLAG(&huart2, USART_ISR_REACK) == RESET);

...

  /*Start LPUART receive on IT*/
  HAL_UART_Receive_IT(&huart2, &charRx, 1);

...

}
```

ALABSTM, 2022-01-30:
Hi @RomanJasmann,

Thank you for this report. We will get back to you with a feedback as soon as possible. Please excuse the delay it may take us sometimes to reply. Thank you for your comprehension.

With regards,

ASELSTM, 2022-02-23:
ST Internal Reference: 123352

acutetech, 2022-12-09:
Slightly off topic, but I found this when searching for vcom_ReceiveInit() 

I am trying to get LoRaWAN_End_Node working with incoming UART characters waking the processor from STOP1 mode. I can't find any examples that I might modify that use vcom_ReceiveInit(). Sounds like RomanJasmann has used vcom_ReceiveInit(). Can he or anyone point me to a working example? Especially if it wakes from STOP1.

ASELSTM, 2023-08-18:
Hi @RomanJasmann,

Sorry for this late answer. Would you please try to use the latest version of the STM32CubeWL **V1.3.0** and check whether you are still having the same issue or not.

With regards,

ASELSTM, 2023-09-04:
Hi @RomanJasmann,

Please allow me to close this thread as no activity. You may reopen it at anytime if you have any details to share with us in order to help you to solve the issue. Thank you for your comprehension.

With regards,

closed 2023-09-04 by ASELSTM