STMicroelectronics/STM32CubeWL #35: External Interrupt hangs STM32WLE5CC when MCU reads SPI EEPROM, or Using I2C

mightChamp, 2022-02-22:
Hello,

I am using STM32WLE5CC with STM32Cube Studio 1.8.0.

I am using M95512 SPI EEPROM, and SSD1306 I2C Oled.

I can communicate with both peripheral and all works perfect.

Our project is of Counting Reed switch pulse from external Meter. Pulse can occur any time.

We faced below 2 issues:

1) when Reading/using SPI or I2C Peripheral, and External interrupt trigger, MCU stucks.

2) when Reset/Reboot occurs and have External Interrupt trigger, then MCU stucks.

MCU stucks in a situation where its recovery is only possible with reset, or watchdog reset. But watchdog is not a solution as it will miss pulse reading count.

We also observed that MCU doesn't stck in Error_hanlder() function.

Also, We are not using SPI/I2C interrupt. We use polling method for SPI/I2C communication.

Please help us as soon as possible.

Thank You

mightChamp, 2022-03-08:
We found that when external interrupt & other function uses APP_PRINTF() function, then MCU get stucks.

In debugger, it is found that, MCU stuck in stm32_tiny_vsnprintf.c

in tiny_vsnprintf_like() function at line 511.

`*str++ = *fmt;`

MCU doesn't go into any error handler, as I have put NVIC_systemReset() in each Error Handler functions shown below, but MCU doesn't get reset.

```
NMI_Handler(),
HardFault_Handler(),
MemManage_Handler(),
BusFault_Handler(),
UsageFault_Handler(),
SVC_Handler(),
DebugMon_Handler(),
PendSV_Handler(),
Error_Handler(),
```

So, I wanted to know,
1) how it can be stuck in tiny_vsnprintf_like() function?
2) why any error handler is not called if it stucks?

Thank You

ASELSTM, 2023-04-10:
Hi @mightChamp,

Would you please try to use the latest tag of STM32CubeWL and check whether you still have the same issue.

With regards,

ASELSTM, 2023-07-11:
Hi @mightChamp,

Please allow me to close this thread as no activity. You may reopen it at anytime if you have any details to share with us in order to help you to solve the issue. Thank you for your comprehension.

With regards,

closed 2023-07-11 by ASELSTM