STMicroelectronics/STM32CubeWL #36: Any plans to update library here to V1.2.0?

KaDw, 2022-03-23:
According to STM32CubeMX release notes this package has V1.2.0 released on 04-February-2022
Could you also update this repository?

smartech-sp, 2022-03-23:
@KaDw 
in the meantime v1.2.0 is available here
https://www.st.com/en/embedded-software/stm32cubewl.html

pkral78, 2022-07-26:
+1 use github repo as uptodate source is just best

KaDw, 2022-09-28:
Any news? There is no point in including this as submodule if you don't update it for over 7 months.

KaDw, 2022-12-08:
> Hi @KaDw,
> 
> Done for this repo. Still need to do it for both [cmsis_device_wl](https://github.com/STMicroelectronics/cmsis_device_wl) and [stm32wlxx_hal_driver](https://github.com/STMicroelectronics/stm32wlxx_hal_driver) repos.
> 
> With regards, 

Thank you!

smartech-sp, 2022-12-08:
> Hi @KaDw,
> 
> Done for this repo. Still need to do it for both [cmsis_device_wl](https://github.com/STMicroelectronics/cmsis_device_wl) and [stm32wlxx_hal_driver](https://github.com/STMicroelectronics/stm32wlxx_hal_driver) repos.
> 
> With regards,

Now? After all this time?
v1.3 came out a few days ago :)