STMicroelectronics/STM32CubeWL #37: Join procedure fails after changing the AppKey

PG430, 2022-04-01:
Hi,

During the testing we have found something interesting. During the Join procedure device is using Network key instead of AppKey
As per the LoRA spec device must use AppKey to generate message integrity code(MIC). 

[image]

[image]

we have found this when we have change the appkey of device and server, after the changes we are seeing MIC failure in join request.
Please help to understand why Network Key is in use for join procedure?

ASELSTM, 2022-04-11:
ST Internal Reference: 126216

PG430, 2022-04-14:
Any comment?

PG430, 2022-05-02:
@ASELSTM any update? anything that you can share will be helpful

PG430, 2022-05-04:
@ASELSTM   Please share any available update regarding this concern.

PG430, 2022-06-15:
@ALABSTM can you share any updates on this issue?

PG430, 2022-07-18:
@ASELSTM can you share any updates on this issue?

ALABSTM, 2022-12-08:
Duplicate of #24.

closed 2022-12-08 by ALABSTM

PG430, 2022-12-09:
@ALABSTM Thank you for the detail clarification.