STMicroelectronics/STM32CubeWL #38: FUOTA on STM32WLE5?

chris-mds, 2022-06-06:
According to the documentation from Application Note AN5554:

_The FUOTA application supports a full firmware upgrade image (entire firmware image sent to the end device) only runs on STM32WL55xx target_

Is this related to the documentation sheet or for FUOTA in general? It is a little bit confusing for me, because STM32WL55xx is DualCore series, but in this repository there is an FUOTA  example application for single core. So FUOTA should work with single core and therefore with STM32WLE5 series?

ASELSTM, 2022-06-24:
ST Internal Reference: 130956

ASELSTM, 2022-11-15:
Hi @chris-mds,

That is right, the FUOTA application is available for both single and dual core. The application note **AN5554** will be updated to clarify the case of STM32WLE5 users.

With regards,

RJMSTM, 2023-09-29:
Hello @chris-mds,

Please allow me to close this thread as no activity. You may reopen it at any time if you have any details to share with us in order to help you to solve the issue. Thank you for your comprehension.

With regards

closed 2023-09-29 by RJMSTM