STMicroelectronics/STM32CubeWL #41: Problems with file names when compiling on Linux

lucasgranberg, 2022-08-09:
I don't know if Linux is officially supported but I got it to work. Only problem was with some lower/upper-case file names. I had to edit these lines to use upper case "PLUS" to get it to work
```bash
  priorbin=$userAppBinary"/SBSFU_LoRaWAN_End_Node_DualCore_CM0PLUS.bin"
  priorbinadd=$userAppBinary"/SBSFU_LoRaWAN_End_Node_DualCore_CM0PLUS.bin.baseadd"
```

https://github.com/STMicroelectronics/STM32CubeWL/blob/59c5722deaf51724cff8446413a7a76792d8bb81/Projects/NUCLEO-WL55JC/Applications/LoRaWAN_FUOTA_DualCore/2_Images_SECoreBin/STM32CubeIDE/SECBOOT_ECCDSA_WITH_AES128_CBC_SHA256.sh#L38

ASELSTM, 2022-11-07:
ST Internal Reference: 138402

RJMSTM, 2025-09-24:
I hope you are fine. The issue you reported has been fixed in the frame of version v1.3.1 of the STM32CubeWL published recently on GitHub. Thank you again for having reported.