STMicroelectronics/STM32CubeWL #49: RxParams.LinkCheck is never cleared

J0hanss0n, 2022-09-12:
**Set-up**
LSM100A based Evaluation Kit (STM32WL55JC)
STM32CubeIDE Version: 1.10.1
https://github.com/SeongJiIoT/LSM100A_SDK

**Description**
After piggy-pack a link check request [RxParams.LinkCheck](https://github.com/STMicroelectronics/STM32CubeWL/blob/59c5722deaf51724cff8446413a7a76792d8bb81/Middlewares/Third_Party/LoRaWAN/LmHandler/LmHandler.c#L1022) is set during rx processing. But this flag is never cleared afterwards.
From then on each time a downlink message is received the DMODM and GWN parameters are printed to serial in [AT_event_receive()](https://github.com/STMicroelectronics/STM32CubeWL/blob/59c5722deaf51724cff8446413a7a76792d8bb81/Projects/NUCLEO-WL55JC/Applications/LoRaWAN/LoRaWAN_AT_Slave/LoRaWAN/App/lora_at.c#L231).

I'd suppose to clear the flag in [LmHandlerProcess()](https://github.com/STMicroelectronics/STM32CubeWL/blob/59c5722deaf51724cff8446413a7a76792d8bb81/Middlewares/Third_Party/LoRaWAN/LmHandler/LmHandler.c#L452) right before the actual processing.

ASELSTM, 2022-09-16:
ST Internal Reference: 134779

ASELSTM, 2023-02-20:
Hi @J0hanss0n,

Would you please try to test with the latest available version of STM32CubeWL and check whether you are still having the same issue or not.

With regards,

ASELSTM, 2023-05-17:
Hi @J0hanss0n,

Please allow me to close this thread as no activity. You may reopen it at anytime if you have any details to share with us in order to help you to solve the issue. Thank you for your comprehension.

With regards,

closed 2023-06-08 by RJMSTM