STMicroelectronics/STM32CubeWL #50: RX event output after Join Request shows wrong DR

J0hanss0n, 2022-09-14:
**Set-up**
LSM100A based Evaluation Kit (STM32WL55JC)
STM32CubeIDE Version: 1.10.1
https://github.com/SeongJiIoT/LSM100A_SDK

**Description**
After ADR changed the data rate for example to DR5 and then issuing a new JOIN request, the wrong DR is shown in the event output:
```
AT+JOIN=1
OTAA mode
1933s038:TX on freq 868500000 Hz at DR 0 and 13 dBm
OK
1934s819:MAC txDone
1939s705:RX_1 on freq 868500000 Hz at DR 0
1941s169:MAC rxDone
+EVT:JOINED
+EVT:RX_1, DR 5, RSSI -85, SNR 5
```
This is because the RxParams.Datarate field is only updated in [McpsIndication()](https://github.com/STMicroelectronics/STM32CubeWL/blob/59c5722deaf51724cff8446413a7a76792d8bb81/Middlewares/Third_Party/LoRaWAN/LmHandler/LmHandler.c#L950) but not in MlmeIndication().

I'd suppose to move the Datarate field from [McpsIndication_t](https://github.com/STMicroelectronics/STM32CubeWL/blob/59c5722deaf51724cff8446413a7a76792d8bb81/Middlewares/Third_Party/LoRaWAN/Mac/LoRaMac.h#L983) to [LoRaMacRxStatus_t](https://github.com/STMicroelectronics/STM32CubeWL/blob/59c5722deaf51724cff8446413a7a76792d8bb81/Middlewares/Third_Party/LoRaWAN/Mac/LoRaMac.h#L320) and use this where applicable and in addition in [MlmeIndication](https://github.com/STMicroelectronics/STM32CubeWL/blob/59c5722deaf51724cff8446413a7a76792d8bb81/Middlewares/Third_Party/LoRaWAN/LmHandler/LmHandler.c#L1084):
```
static void MlmeIndication( MlmeIndication_t *mlmeIndication, LoRaMacRxStatus_t *RxStatus )
{
    RxParams.IsMcpsIndication = 0;
    RxParams.Status = mlmeIndication->Status;
    RxParams.Datarate = RxStatus->RxDatarate;
    RxParams.Rssi = RxStatus->Rssi;
    RxParams.Snr = RxStatus->Snr;
    ...
```

ASELSTM, 2022-09-16:
ST Internal Reference: 134767

ASELSTM, 2022-09-16:
Hi @J0hanss0n,

This bug have been fixed in the version **v2.4.0** of customized LoRaWAN version for STM32Cube solution provided within the STM32CubeWL **v1.2.0** available on [ST.com](https://www.st.com/en/embedded-software/stm32cubewl.html#get-software). 

The proposed fix to prevent a JoinAccept with uninitialized fields in RxStatus output parameter is as follow. 

```diff
static void MlmeIndication( MlmeIndication_t *mlmeIndication, LoRaMacRxStatus_t
 {
     RxParams.IsMcpsIndication = 0;
     RxParams.Status = mlmeIndication->Status;
+    RxParams.Datarate = mlmeIndication->RxDatarate;
     RxParams.Rssi = rxStatus->Rssi;
     RxParams.Snr = rxStatus->Snr;
     RxParams.RxSlot = rxStatus->RxSlot;
+    RxParams.DownlinkCounter = mlmeIndication->DownLinkCounter;
     if ((mlmeIndication->MlmeIndication != MLME_BEACON) && (mlmeIndication->MlmeIndication != MLME_BEACON_LOST))
     {
         LmHandlerCallbacks->OnRxData( NULL, &RxParams );
```

```diff
static void ProcessRadioRxDone( void )
                     MacCtx.McpsIndication.Status = LORAMAC_EVENT_INFO_STATUS_ERROR;
                 }
                 MacCtx.McpsIndication.DownLinkCounter = downLinkCounter;
+                MacCtx.MlmeIndication.DownLinkCounter = downLinkCounter;
                 PrepareRxDoneAbort( );
                 return;
             }
```
```diff
static void ProcessRadioRxDone( void )
                     MacCtx.McpsIndication.Status = LORAMAC_EVENT_INFO_STATUS_ERROR;
                 }
                 MacCtx.McpsIndication.DownLinkCounter = downLinkCounter;
+                MacCtx.MlmeIndication.DownLinkCounter = downLinkCounter;
                 PrepareRxDoneAbort( );
                 return;
             }
```
```diff
static void ProcessRadioRxDone( void )
             MacCtx.McpsIndication.Buffer = NULL;
             MacCtx.McpsIndication.BufferSize = 0;
             MacCtx.McpsIndication.DownLinkCounter = downLinkCounter;
+            MacCtx.MlmeIndication.DownLinkCounter = downLinkCounter;
             MacCtx.McpsIndication.AckReceived = macMsgData.FHDR.FCtrl.Bits.Ack;
 
             MacCtx.McpsConfirm.Status = LORAMAC_EVENT_INFO_STATUS_OK;
```
```diff
static void RxWindowSetup( TimerEvent_t* rxTimer, RxConfigParams_t* rxConfig )
 
     if( RegionRxConfig( Nvm.MacGroup2.Region, rxConfig, ( int8_t* )&MacCtx.McpsIndication.RxDatarate ) == true )
     {
+        MacCtx.MlmeIndication.RxDatarate = MacCtx.McpsIndication.RxDatarate;
         Radio.Rx( Nvm.MacGroup2.MacParams.MaxRxWindow );
         MacCtx.RxSlot = rxConfig->RxSlot;
     }
```
```diff
static void OpenContinuousRxCWindow( void )
     // Thus, there is no need to set the radio in standby mode.
     if( RegionRxConfig( Nvm.MacGroup2.Region, &MacCtx.RxWindowCConfig, ( int8_t* )&MacCtx.McpsIndication.RxDatarate ) == true )
     {
+        MacCtx.MlmeIndication.RxDatarate = MacCtx.McpsIndication.RxDatarate;
         Radio.Rx( 0 ); // Continuous mode
         MacCtx.RxSlot = MacCtx.RxWindowCConfig.RxSlot;
     }
```

```diff
typedef struct sMlmeIndication
      * Status of the operation
      */
     LoRaMacEventInfoStatus_t Status;
+    /*!
+     * Downlink datarate
+     */
+    uint8_t RxDatarate;
+    /*!
+     * The downlink counter value for the received frame
+     */
+    uint32_t DownLinkCounter;
     /*!
      * Beacon information. Only valid for \ref MLME_BEACON,
      * status \ref LORAMAC_EVENT_INFO_STATUS_BEACON_LOCKED
```

As the issue is already fixed please allow me to close this thread.  Thank you for your contribution.

With regards,

closed 2022-09-16 by ASELSTM