STMicroelectronics/STM32CubeWL #51: Rx Timeout on join when using the LSI clock

rayaanlatief, 2022-09-29:
We have a custom PCB with the stm32wl55 MCU, and for the most part, the reference design was followed. However the LSE crystal was not added. 

I have updated the example code and enabled the LSI clock source. 

For my application, we are using OTAA.

The join request is received by the gateway and responds to it. However, my end node does not receive the response.

I have tested with the nucleo-wl55 dev board with the LSI clock and disable the LSE clock source the device fails to join, However once enable the LSE clock source the device joins succesfully

IDE used: STM32CubeIDE 1.10.1

RomanJasmann, 2022-10-10:
@rayaanlatief 

1. Have you configured the LSI clock to have a tick period below 1 ms?
2. Have you adapted TIMER_IF_Convert_Tick2ms, TIMER_IF_Convert_ms2Tick and TIMER_IF_GetTime to reflect the LSI clock frequency?
3. Consider that the LSI RC oscillator can have a high frequency error.

[image]

HBOSTM, 2023-04-14:
Hello @rayaanlatief ,

Thank you for this contribution.
I can't reproduce the problem from my side. So could you please give us more details about how you got this issue? And share a demo of the code you have used to reproduce this issue.

Best Regards,

HBOSTM, 2023-05-04:
Hello @rayaanlatief,

Please allow me to close this thread as no activity. You may reopen it at anytime if you have any details to share with us in order to help you to solve the issue. Thank you for your comprehension.

With regards,

closed 2023-05-04 by HBOSTM