STMicroelectronics/STM32CubeWL #53: Initialize loraInfo structure to zero

KaDw, 2022-11-13:
Setup:

LoRa-E5-STM32WLE5JC
GNU Arm Embedded Toolchain 10.3-2021.10

Suggestion:
Minor thing I noticed. In lora_info_template.c and all of the lora_info.c files in examples, part of `loraInfo` struct is uninitialized.
According to the struct declaration it has 4 members but when it is created it has only two first members set to zero:
`static LoraInfo_t loraInfo = {0, 0};`

However, all members are set to zero in Init function (I think the structure isn't accessed before this init). This makes my suggestion really cosmetic thing.
```
void LoraInfo_Init(void)
{
  loraInfo.ActivationMode = 0;
  loraInfo.Region = 0;
  loraInfo.ClassB = 0;
  loraInfo.Kms = 0;
...
```

Solution:
`static LoraInfo_t loraInfo = {0};`

ASELSTM, 2022-11-21:
ST Internal Reference: 139396

ASELSTM, 2023-02-20:
Hi @KaDw,

The issue you reported has been fixed in the frame of version v1.2.0 of the STM32CubeWL published on GitHub.

Thank you again for having reported.

With regards,

closed 2023-02-20 by ASELSTM