STMicroelectronics/STM32CubeWL #54: Compilation error with USE_LRWAN_1_1_X_CRYPTO set to 1

KaDw, 2022-11-15:
Setup:

LoRa-E5-STM32WLE5JC
GNU Arm Embedded Toolchain 10.3-2021.10

Bug:

USE_LRWAN_1_1_X_CRYPTO is not recognized in LoRaMacTypes.h and evaluates to zero. eKeyIdentifier enum values are always the same. This is not the only error when I set USE_LRWAN_1_1_X_CRYPTO
Setting `#define USE_LRWAN_1_1_X_CRYPTO` in LoRaMacCrytpo.h to 1 results in following error: 
```
In file included from Middlewares/Third_Party/LoRaWAN/Crypto/soft-se.c:55:
LoRaWAN/App/se-identity.h:175:25: error: 'J_S_INT_KEY' undeclared here (not in a function)
  175 |             .KeyID    = J_S_INT_KEY,                                                                                \
      |                         ^~~~~~~~~~~
LoRaWAN/App/se-identity.h:383:9: note: in expansion of macro 'SESSION_KEYS_LIST'
  383 |         SESSION_KEYS_LIST                                                                                           \
      |         ^~~~~~~~~~~~~~~~~
Middlewares/Third_Party/LoRaWAN/Crypto/soft-se.c:118:14: note: in expansion of macro 'SOFT_SE_KEY_LIST'
  118 |   .KeyList = SOFT_SE_KEY_LIST
      |              ^~~~~~~~~~~~~~~~
LoRaWAN/App/se-identity.h:184:25: error: 'J_S_ENC_KEY' undeclared here (not in a function)
  184 |             .KeyID    = J_S_ENC_KEY,                                                                                \
      |                         ^~~~~~~~~~~
LoRaWAN/App/se-identity.h:383:9: note: in expansion of macro 'SESSION_KEYS_LIST'
  383 |         SESSION_KEYS_LIST                                                                                           \
      |         ^~~~~~~~~~~~~~~~~
Middlewares/Third_Party/LoRaWAN/Crypto/soft-se.c:118:14: note: in expansion of macro 'SOFT_SE_KEY_LIST'
  118 |   .KeyList = SOFT_SE_KEY_LIST
      |              ^~~~~~~~~~~~~~~~
LoRaWAN/App/se-identity.h:193:25: error: 'F_NWK_S_INT_KEY' undeclared here (not in a function)
  193 |             .KeyID    = F_NWK_S_INT_KEY,                                                                            \
      |                         ^~~~~~~~~~~~~~~
LoRaWAN/App/se-identity.h:383:9: note: in expansion of macro 'SESSION_KEYS_LIST'
  383 |         SESSION_KEYS_LIST                                                                                           \
      |         ^~~~~~~~~~~~~~~~~
Middlewares/Third_Party/LoRaWAN/Crypto/soft-se.c:118:14: note: in expansion of macro 'SOFT_SE_KEY_LIST'
  118 |   .KeyList = SOFT_SE_KEY_LIST
      |              ^~~~~~~~~~~~~~~~
LoRaWAN/App/se-identity.h:201:25: error: 'S_NWK_S_INT_KEY' undeclared here (not in a function)
  201 |             .KeyID    = S_NWK_S_INT_KEY,                                                                            \
      |                         ^~~~~~~~~~~~~~~
LoRaWAN/App/se-identity.h:383:9: note: in expansion of macro 'SESSION_KEYS_LIST'
  383 |         SESSION_KEYS_LIST                                                                                           \
      |         ^~~~~~~~~~~~~~~~~
Middlewares/Third_Party/LoRaWAN/Crypto/soft-se.c:118:14: note: in expansion of macro 'SOFT_SE_KEY_LIST'
  118 |   .KeyList = SOFT_SE_KEY_LIST
      |              ^~~~~~~~~~~~~~~~
LoRaWAN/App/se-identity.h:210:25: error: 'NWK_S_ENC_KEY' undeclared here (not in a function); did you mean 'NWK_S_KEY'?
  210 |             .KeyID    = NWK_S_ENC_KEY,                                                                              \
      |                         ^~~~~~~~~~~~~
LoRaWAN/App/se-identity.h:383:9: note: in expansion of macro 'SESSION_KEYS_LIST'
  383 |         SESSION_KEYS_LIST                                                                                           \
      |         ^~~~~~~~~~~~~~~~~
Middlewares/Third_Party/LoRaWAN/Crypto/soft-se.c:118:14: note: in expansion of macro 'SOFT_SE_KEY_LIST'
  118 |   .KeyList = SOFT_SE_KEY_LIST
      |              ^~~~~~~~~~~~~~~~
```

Fix:
Remove #if USE_LRWAN_1_1_X_CRYPTO statements from the eKeyIdentifier enum.

ASELSTM, 2022-11-21:
ST Internal Reference: 139386

ASELSTM, 2023-02-20:
Hi @KaDw,

As `USE_LRWAN_1_1_X_CRYPTO` has been removed form the STM32CubeWL V1.2.0 and V1.3.0 recently published, please  allow me thus to close this thread.

With regards,

closed 2023-02-20 by ASELSTM