STMicroelectronics/STM32CubeWL #56: RTC Accuracy issue

PG430, 2022-12-08:
[@ALABSTM] (https://github.com/STMicroelectronics/STM32CubeH7/issues/url) we are working on new release of STM32WL "STM32Cube_FW_WL_V1.3.0". release has introduced changes in RTC driver which we believe is fixed for the issue (RTC loosing time issue https://github.com/STMicroelectronics/STM32CubeH7/issues/24 )
But, after running RTC overnight we have seen RTC issue is still persist in different manner this time RTC is increasing time

Test Start time : 17:55:56, UTC time 1670415956
Test End Time: 9:48:00, UTC time 1670473080, (57124 Seconds)
Difference of UTC time should be : 57124 (1670473080 - 1670415956)

But, From attached logs we can see Test End time is 9:48:00 and total seconds are 57135s837 (check last line in logs).
RTC had generated 57135s837 seconds instead of 57124 Seconds.

So, 57135 - 57124 = 11 Seconds are increased in actual time

[RTCTestLogs.txt](https://github.com/STMicroelectronics/STM32CubeH7/files/10182315/RTCTestLogs.txt)
[RTC Driver.zip](https://github.com/STMicroelectronics/STM32CubeH7/files/10182344/RTC.Driver.zip)
Driver files are also attached for more details

Can you please explain what exactly is happening after RTC fix is applied? how to fix this?

PG430, 2022-12-20:
Hello @ALABSTM,

Thank you for the quick reply. Any finding on this issue? if not, is it possible for you to take this on priority? we require RTC in multiple projects and we are stuck due to this problem. Appreciate your support on this matter.

kaunomedis, 2023-01-08:
This bug was detected **MANY years ago** and no official solution.

qkrwogur, 2023-03-17:
I hope you can solve this problem soon. If the time doesn't match, it's not a timer.  Come up with a solution quickly!

ALABSTM, 2023-06-27:
Hi @PG430,

First of all, please excuse this late reply.

The fix applied prevents from entering the _initialization_ mode when executing `HAL_RTC_Init()` function in case the RTC is already initialized. Indeed, while in the _initialization_ mode, the RTC's calendar (**i.e.**, time and date) are **frozen**, causing the loss of seconds.

Regarding your case, I might advise the following, if not yet done:
* It is better to use the LSE clock as input signal than to use the LSI clock.
* The LSE clock frequency is generally **32,768 Hz**. To obtain a **1 Hz** frequency at the output, you shall ensure that RTC's prescalers values are:
  * 0x7F for the asynchronous prescaler
  * 0xFF for the synchronous prescaler

Now, as nothing is perfect, the RTC offers "fine-tuning" mechanisms, mainly:
* The smooth digital calibration,
* The synchronization with a more accurate clock.

Both mechanisms and more are presented in the [AN4759](https://www.st.com/resource/en/application_note/an4759-using-the-hardware-realtime-clock-rtc-and-the-tamper-management-unit-tamp-with-stm32-microcontrollers-stmicroelectronics.pdf) application note.

I hope this helps. Please do not hesitate otherwise.

With regards,

ALABSTM, 2023-07-20:
Hi @PG430,

I hope you are fine. I also hope the clues provided in my previous comment helped you get the accuracy you are looking for. Please allow me to close this thread.

With regards,

closed 2023-07-20 by ALABSTM