STMicroelectronics/STM32CubeWL #60: NVM DATA RESTORED not working with STM32WLE5xB (fix included)

Yoshi20, 2023-01-20:
I discovered that NVM_DATA_RESTORED is never shown after the context was saved (AT+CS) when using the AT slave firmware from STM32CubeWL_V1.3.0 on a STM32WLE5CB.

After (too many) hours, I discovered that in LoRaWAN/App/lora_at.c the LORAWAN_NVM_BASE_ADDRESS define is wrong for this microcontroller.

Workaround:
```
#define LORAWAN_NVM_BASE_ADDRESS                    ((void *)0x0803F000UL)

/* USER CODE BEGIN PD */
#undef LORAWAN_NVM_BASE_ADDRESS
// fix bug from ST => define addr 124k and not 252k for this uC (with 128k flash)
#define LORAWAN_NVM_BASE_ADDRESS                    ((void *)0x0801F000UL)
/* USER CODE END PD */
```

RJMSTM, 2023-03-06:
Hello @Yoshi20 

Thank you for your contribution. Would it be possible to share the reference of the datasheet or Application note.

BeST Regards,
Rania

RJMSTM, 2023-11-28:
Hello,
Any update please

RJMSTM, 2023-11-28:
Hello,
Please allow me to close this thread as no activity. You may reopen it at any time if you have any details to share with us in order to help you to solve the issue. Thank you for your comprehension.

With regards