STMicroelectronics/STM32CubeWL #63: Problem with using uart1 SubGHz_Phy_Per and STM32WL-SubGHz-PhyBasic-Tx-Rx examples

Javadbolboli, 2023-03-13:
Hello everyone,

I am encountering an issue with UART1 while using STM32WL for LoRa Communication with STM32WL-SubGHz-PhyBasic-Tx-Rx and SubGHz_Phy_Per Apps (Applications/SubGHz_Phy/SubGHz_Phy_Per).
In my application, I receive a message from LoRa  and intend to transmit it via UART1 which is set to PA9 and PA10. However, when I check the UART RX using a logic analyzer, I do not receive anything.

I have tested the UART1 alone in a simple project, and it works fine. But, when I integrate UART1 into the SubGHz_Phy_Per and STM32WL-SubGHz-PhyBasic-Tx-Rx examples, it does not work.

I have added HAL-UART to the OnTxDone function, main , Rx_Process and Per_Processand , but the issue persists. Can someone please suggest a solution to this problem?

Javadbolboli, 2023-03-15:
Update on the problem:
I have successfully solved the issue with the STM32WL-SubGHz-PhyBasic-Tx-Rx examples by noticing that the UART1 clock was not active while i activated it before. After importing the project again, the project worked properly.

However, the issue still persists with the SubGHz_Phy_Per example. Interestingly, I found that the UART1 works fine when I am in debug mode, but not when I'm not in debug mode. I even added some hal-delay to check if the problem was due to timing, but there was no change.

If you have any additional insights or suggestions, they would be greatly appreciated.

ASELSTM, 2023-07-11:
Hi @Javadbolboli,

First, excuse this late answer, hope you have solved you issue since then. 
Would you please make sure you are correctly initializing the UART1 peripheral clock before enabling the UART1 clock.

With regards,

ASELSTM, 2023-07-31:
Hi @Javadbolboli,

Please allow me to close this thread as no activity. You may reopen it at anytime if you have any details to share with us in order to help you to solve the issue. Thank you for your comprehension.

With regards,

closed 2023-07-31 by ASELSTM

Javadbolboli, 2023-07-31:
Hi
Sure, but I'm on vacation
 I will do it after tomorrow
Thanks

Javadbolboli, 2023-08-02:
Hi, I this it is closed now