STMicroelectronics/STM32CubeWL #64: `SUBGRF_SetRfTxPower` interprets `power` argument incorrectly.

Miceuz, 2023-03-24:
The bug is in `Middlewares/Third_Party/SubGHz_Phy/stm32_radio_driver/radio_driver.h`:

`SUBGRF_SetRfTxPower` expects `Tx power level [0..15]`. It does checks on `power > 15` to select `RFO_HP`  and then it calls `SUBGRF_SetTxParams` that expects `RF output power [-18..13] dBm`.

So it interprets `power` argument incorrectly as it it's index, not dBm while later it's used as dBm. All the client code in https://github.com/stm32duino/STM32LoRaWAN/ uses `SUBGRF_SetRfTxPower` providing power in dBm.

RJMSTM, 2023-03-28:
Hi @Miceuz,

Thank you for this report. We will get back to you as soon as we analyze it further. This may take some time. Thank you for your comprehension.

With regards,
Rania

fpistm, 2023-04-11:
Hi @Miceuz 
and thanks to submit this issue.
@RJMSTM is there an internal BZ about this? Thanks.