STMicroelectronics/STM32CubeWL #66: SUBGHZ HAL driver using SPI polling and not DMA... too slow

pieterconradie, 2023-03-28:
Hi there!

I have measured the SPI comms using a logic analyzer:
[image]

As you can see there is plenty room for improvement using DMA instead of polling. I'm referring to stm32wlxx_hal_subghz.c SUBGHZSPI_Transmit() and SUBGHZSPI_Receive():
https://github.com/STMicroelectronics/STM32CubeWL/blob/main/Drivers/STM32WLxx_HAL_Driver/Src/stm32wlxx_hal_subghz.c#L1626-L1735

It would be better to formulate the SPI packet for transfer and then use DMA. This means that a HAL_SUBGHZ_ReadRegisters() would have to be rewritten:
https://github.com/STMicroelectronics/STM32CubeWL/blob/main/Drivers/STM32WLxx_HAL_Driver/Src/stm32wlxx_hal_subghz.c#L863-L917

These unnecessary delays between each SPI byte will impact battery life. It is most notable when transmitting or receiving a large packet.

Best regards,
Pieter
https://piconomix.com

RJMSTM, 2023-03-28:
hello @pieterconradie 

Thank you for this report. We will get back to you as soon as we analyze it further. This may take some time. Thank you for your comprehension.

With regards,
Rania

KRASTM, 2024-09-10:
ST internal Reference: 190600

pieterconradie, 2024-09-10:
Here is an example implementation for free :). For small SPI packets regular polling is used and for large SPI packets DMA is used.

[stm32wlxx_hal_subghz.c.log](https://github.com/user-attachments/files/16942387/stm32wlxx_hal_subghz.c.log)

KRASTM, 2024-10-04:
Hello @pieterconradie,

Regarding your proposal, I concur with your recommendation, but unfortunately, there is no plan to update the driver. I will keep you informed if there are any updates. Please allow me to close this issue.

Thank you in advance for your comprehension.

With regards,

pieterconradie, 2024-10-04:
Yes you are welcome to close the issue.

Regards,
Pieter

closed 2024-10-04 by KRASTM