STMicroelectronics/STM32CubeWL #68: LPUART is not working inside Seeed-LoRa-E5_PingPong

crkco, 2023-04-06:
I have two STM32WL5* boards that sucessfully used the SubGHz_Phy_PingPong example.

For additional functionality I added the typical routine for implementing LPUART as follows:
```

static void MX_LPUART1_UART_Init(void)
{

  /* USER CODE BEGIN LPUART1_Init 0 */

  /* USER CODE END LPUART1_Init 0 */

  /* USER CODE BEGIN LPUART1_Init 1 */

  /* USER CODE END LPUART1_Init 1 */
  hlpuart1.Instance = LPUART1;
  hlpuart1.Init.BaudRate = 9600;
  hlpuart1.Init.WordLength = UART_WORDLENGTH_8B;
  hlpuart1.Init.StopBits = UART_STOPBITS_1;
  hlpuart1.Init.Parity = UART_PARITY_NONE;
  hlpuart1.Init.Mode = UART_MODE_TX_RX;
  hlpuart1.Init.HwFlowCtl = UART_HWCONTROL_NONE;
  hlpuart1.Init.OneBitSampling = UART_ONE_BIT_SAMPLE_DISABLE;
  hlpuart1.Init.ClockPrescaler = UART_PRESCALER_DIV1;
  hlpuart1.AdvancedInit.AdvFeatureInit = UART_ADVFEATURE_NO_INIT;
  hlpuart1.FifoMode = UART_FIFOMODE_DISABLE;
  if (HAL_UART_Init(&hlpuart1) != HAL_OK) //RETURNS HAL_ERROR
  {
    Error_Handler();
  }
  if (HAL_UARTEx_SetTxFifoThreshold(&hlpuart1, UART_TXFIFO_THRESHOLD_1_8) != HAL_OK)
  {
    Error_Handler();
  }
  if (HAL_UARTEx_SetRxFifoThreshold(&hlpuart1, UART_RXFIFO_THRESHOLD_1_8) != HAL_OK)
  {
    Error_Handler();
  }
  if (HAL_UARTEx_DisableFifoMode(&hlpuart1) != HAL_OK)
  {
    Error_Handler();
  }*/
  /* USER CODE BEGIN LPUART1_Init 2 */

  /* USER CODE END LPUART1_Init 2 */

}

```

When I run this code for a newly created project with CubeMX it works flawlessly.

However, when I try to use it in your example, the `HAL_UART_Init(&hlpuart1)` returns HAL_ERROR and is therefore not usable.
Similarly, any other uart doesn't work either.

RJMSTM, 2023-04-20:
Hi @crkco,

Thank you for this report. We will get back to you as soon as we analyze it further. This may take some time. Thank you for your comprehension.

With regards,

ASELSTM, 2023-05-16:
Hi @crkco,

I guess the issue you are facing is related to the chosen clock prerescaler value. In fact, in the LPUART case, fCK must range from 
  **3 x baudrate** to **4096 x baudrate**,  a check is thus performed in the `UART_SetConfig()` to verify this value.

https://github.com/STMicroelectronics/STM32CubeWL/blob/8a8140f342a3d0ac85e5a02f44db8782765a7bed/Drivers/STM32WLxx_HAL_Driver/Src/stm32wlxx_hal_uart.c#L3208-L3215

As in your case the baudrate is set to **9600**,  would you thus try to make sure that the frequency clock value is well ranged between 3* 9600 and  4096 * 9600. For that you can change the `hlpuart1.Init.ClockPrescaler` value.

With regards,

ASELSTM, 2023-06-30:
Hi @crkco,

Please allow me to close this thread as no activity. You may reopen it at anytime if you have any details to share with us in order to help you to solve the issue. Thank you for your comprehension.

With regards,

closed 2023-06-30 by ASELSTM