STMicroelectronics/STM32CubeWL #72: LoRaWAN Timer does not work with RTC_BINARY_MIX

Jakub-Nagy, 2023-05-28:
My code is based off of the LoRaWAN_End_Node example. I had to change the RTC_N_PREDIV_S to 8 because of severe clock drift issues. I also changed the implementation of some of the functions in timer_if.c. Either way, both LoRa and HAL_GetTick, HAL_Delay functions work well.

Now I also need my application to generate a unix timestamp, meaning I need to have the RTC run in BCD mode. So, I set `hrtc.Init.BinMode = RTC_BINARY_MIX` to allow both binary and BCD operation. With this change, the set/get of the calendar works fine, as well as the HAL_Delay and HAL_GetTick build on top of the UTIL_TIMER. But the timer inside the LoRaWAN Middleware stops working - the timeout for RX1 window is opened and never closes.

When I set `hrtc.Init.BinMode = RTC_BINARY_ONLY` the LoRaWAN timers work, but the calendar of course doesn't. What am I missing? Shouldn't RTC_BINARY_MIX allow for operation of both? Thanks.

Jakub-Nagy, 2023-05-29:
Fixed by setting `sAlarm.AlarmMask = RTC_ALARMMASK_ALL` in function `TIMER_IF_StartTimer()`.

closed 2023-05-29 by Jakub-Nagy

harshahegde17, 2025-01-24:
@Jakub-Nagy im also facing the similar issue now, when I use RTC_BINARY_MIX I'm getting RX_TX_timeout issue, please let me know if anymore changes required here.