STMicroelectronics/STM32CubeWL #73: RF Transceiver doesn't support multihopping

EmbeddedAyoub, 2023-06-04:
Hello,
I checked in the reference manual of the mcu and found that it doesn't support the Multihopping channel strategy.
Am I mistaken?
Thanks,
Ayoub

EmbeddedAyoub, 2023-06-05:
> Hello,
>
> Can you specify which reference manual you checked?
>
> With Regards,

Hello,
Yes Sure!
It is  RM0453, the reference manual of the mcu.
What I am trying to do is to make a communication between 2 STM32WL. I'll
send 3 frames in 3 different Physical Channels and Receive them by the
second device.
Thanks,
Ayoub

KRASTM, 2024-10-22:
Hello @EmbeddedAyoub,

Sorry for the late response, regarding your question, we believe there is an application called `SubGHz_Phy_LrFhss` under `\Projects\NUCLEO-WL55JC\Applications\SubGHz_Phy\SubGHz_Phy_LrFhss,` which is for fast frequency hopping spread spectrum (FHSS), and is also mentioned in [AN5406](https://www.st.com/resource/en/application_note/an5406-how-to-build-a-lora-application-with-stm32cubewl-stmicroelectronics.pdf).

I hope this information is helpful to you.

With regards,

KRASTM, 2024-11-05:
@EmbeddedAyoub,

Since there is no activity in a while, please allow me to close this thread. You may reopen it at any time.
Thank you for your comprehension.

With regards,

closed 2024-11-05 by KRASTM