STMicroelectronics/STM32CubeWL #74: SecureElementSetDevEui is not working

karthiclodpi, 2023-06-11:
Hi

We are trying to set dynamic DevEUI and AppKey for OTAA, as well for ABP Devices we are trying to set DevAddr and NWKSKEY and APPSKEY using  SecureElementSetDevEui( devEui ), SecureElementSetKey( APP_KEY,  appkey ) , SecureElementSetDevAddr( ActivationType_t mode, uint32_t devAddr ) etc, before calling LmHandlerConfigure(&LmHandlerParams)

flow :

1) Get DevEUI , AppKey from external controller through UART
2) Using SecureElementSet* functions to set the values, before calling LmHandlerConfigure(&LmHandlerParams)

https://github.com/STMicroelectronics/STM32CubeWL/blob/main/Projects/NUCLEO-WL55JC/Applications/LoRaWAN/LoRaWAN_End_Node/LoRaWAN/App/lora_app.c#L414

Issue :

DevEUI or  AppKey or DevAddr  which are set using SecureElementSet* is not picking up. WLE5 is still using the Default MUC Unique ID.
 
 
 version : 1.3.0
 Controller : STM32WLE5

karthiclodpi, 2023-06-14:
Found solution 

closing issue

closed 2023-06-14 by karthiclodpi

ALABSTM, 2023-07-07:
Hi @karthiclodpi,

Would you mind describing, even briefly, the solution you found so that user who stumble on the same issue you faced could more easily overcome it if they fall on this discussion?

Thank you,