STMicroelectronics/STM32CubeWL #75: Fragdecoder not working as expected sometimes.

nilesh-dryad, 2023-08-30:
**set-up**

Linux Machine Simulator for Frag Decoder. (I have made to make testing faster)
GCC

**bug**

We are encountering difficulties while utilizing the Frag Decoder module. Specifically, we are facing issues with the reconstruction of certain unique sequences of missing fragments. Although the decoder manages to reconstruct all fragments, a subsequent CRC check of the resultant file yields an incorrect value. Upon further investigation, it has become evident that the reconstructed fragments are not being reconstituted accurately.

Here are some details regarding our situation:

We are attempting to send a single file with a size of 499 bytes. The original file has been attached as 'original.txt'. The coded fragments have been stored in a file named 'coded.txt'. Both are binary files but I renamed them as text files. 

During a simulation, we purposefully dropped a selection of uncoded and coded fragments as follows:
Uncoded Fragments (16 in total): 1, 2, 3, 4, 7, 15, 17, 18, 20, 21, 22, 24, 28, 29, 31, 32
Coded Fragments (14 in total):    3, 6, 7, 9, 11, 12, 18, 20, 22, 23, 24, 25, 26, 27

For example, during the transmission of the original 32 fragments, the fragments with the numbers 1, 2, 3, 4, 7, 15, 17, 18, 20, 21, 22, 24, 28, 29, 31, and 32 were not received by the recipient and the same for coded fragments with above 14 indexes.

Our current configuration is as follows:
FRAG_MAX_REDUNDANCY: 1000
FragDecoder->FragNb: 32
FragDecoder->FragSize: 16
codeRate: 32
fragtolerance: 32
fileSize: 499
file CRC: 901076938
number of missing uncoded fragments : 16
number of missing coded fragments : 14

Please see below log of Frag Decoded for more information. You will see FRx print in the log. Its representation is as follow.
```
FRx : a-b/c, d/(e+f)
a = Fragment number of the Fragment received
b = Total number of the Fragment reconstructed by decoder
c = Total number of packet lost at a time of receiving this Fragment
d = Total number of Fragment received at a time of receiving this Fragment
e = total number of expected uncoded fragments
f = total number of expected coded fragments
```

**Decoder Log**

> -- 1693391003 :FRx: 5-0/4, 1/(32+32)
> Missing Indexes:  1  2  3  4  0  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1 
> FragNbRx:5
> FragNbLost:4
> FragNbLastRx:5
> 
> -- 1693391003 :FRx: 6-0/4, 2/(32+32)
> Missing Indexes:  1  2  3  4  0  0  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1 
> FragNbRx:6
> FragNbLost:4
> FragNbLastRx:6
> 
> -- 1693391003 :FRx: 8-0/5, 3/(32+32)
> Missing Indexes:  1  2  3  4  0  0  5  0  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1 
> FragNbRx:8
> FragNbLost:5
> FragNbLastRx:8
> 
> -- 1693391003 :FRx: 9-0/5, 4/(32+32)
> Missing Indexes:  1  2  3  4  0  0  5  0  0  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1 
> FragNbRx:9
> FragNbLost:5
> FragNbLastRx:9
> 
> -- 1693391003 :FRx: 10-0/5, 5/(32+32)
> Missing Indexes:  1  2  3  4  0  0  5  0  0  0  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1 
> FragNbRx:10
[... 222 lines cut ...]
> expected: FE 84 80 8A 51 02 A0 8A 95 58 87 BE 80 5B 88 0A 
> 
> 
> Packet 1 successfully retrieved
> -- 1693391003 :FRx: 64-16/16, 34/(32+32)
> Missing Indexes:  1  2  3  4  0  0  5  0  0  0  0  0  0  0  6  0  7  8  0  9 10 11  0 12  0  0  0 13 14  0 15 16 
> FragNbRx:64
> FragNbLost:16
> FragNbLastRx:33
> 
> -- 1693391003 :dec complete (reconstruct 16 packets)
> -- 1693391003 :dec DONE.
> -- 1693391003 :File CRC ERR Check 959524711 | Orig 901076938
> 

Please note that If I change just one missing index , it works. 
e.g. look at below arrays. only bold index is changed. It was 7 and I made it 8 and it works.
Uncoded Fragments (16 in total): 1, 2, 3, 4, **8**, 15, 17, 18, 20, 21, 22, 24, 28, 29, 31, 32
Coded Fragments (14 in total):    3, 6, 7, 9, 11, 12, 18, 20, 22, 23, 24, 25, 26, 27

For most of the sequence it works, only fews are not working.

**Screenshots**
[coded.txt](https://github.com/STMicroelectronics/STM32CubeWL/files/12474648/coded.txt)
[origin.txt](https://github.com/STMicroelectronics/STM32CubeWL/files/12474649/origin.txt)

KRASTM, 2026-04-03:
ST Internal Reference: CDM0061342

KRASTM, 2026-11-12:
Hello @nilesh-dryad,

Could you please, share which CubeWL package version was used?

With regards,

nilesh-dryad, 2026-11-12:
@KRASTM 
It has been more than 4 years. I think we have resolved it somehow. You can close this.

closed 2026-11-16 by KRASTM