STMicroelectronics/STM32CubeWL #76: Support FreeRTOS with SBSFU

hoai265, 2023-09-15:
Hi guys,
We already have LoRaWAN_SBSFU_1_Slot_DualCore example, could you add LoRaWAN_SBSFU_1_Slot_DualCore what support FreeRTOS or add a manual to support FreeRTOS Lorawan App with SBSFU?

RJMSTM, 2023-11-28:
Hello,
Could you please clarify your question, you want to support FreeRTOS in LoRaWAN_SBSFU_1_Slot_DualCore example?
Thanks

hoai265, 2024-01-18:
Hi @RJMSTM 
Yes, currently, FreeRTOS  Lorawan App examples not work with SBSFU. So i want you guys support FreeRTOS in LoRaWAN_SBSFU_1_Slot_DualCore or others SBSFU examples.
Thanks.

RJMSTM, 2025-06-17:
Hello,

Currently, there are no plans to support FreeRTOS in LoRaWAN_SBSFU_1_Slot_DualCore or any other SBSFU examples.

closed 2025-06-17 by RJMSTM