STMicroelectronics/STM32CubeWL #80: Application hangs because MBMUXIF_LoraSendCmd() command stuck

metaTinker, 2024-03-12:
**Caution**

The Issues are strictly limited for the reporting of problem encountered with the software provided in this project.
For any other problem related to the STM32 product, the performance, the hardware characteristics and boards, the tools the environment in general, please post your report to the **ST Community** in the STM32 MCUs dedicated [page](https://community.st.com/s/topic/0TO0X000000BSqSWAW/stm32-mcus).

**Setup**

 * STM32WL55JC1 embedded on a custom PCB with other peripherals
 *STM32CubeIDE

**Application hangs because MBMUXIF_LoraSendCmd() command stuck on Sem_MbLoRaRespRcv sometimes**

I have an application built around LoRaWAN_End_Node_DualCoreFreeRTOS example provided in the firmware. My application on CM4 sends telemetry roughly every 4-5 minutes. It will run well for a few days and suddenly the MBMUXIF_LoraSendCmd() gets stuck waiting on Sem_MbLoRaRespRcv. Reading more on how dual-core system works I figured that if a response is not received through the IPCC channels, the semaphore is never released. This is a potential pitfall for me because my application requires telemetry to be sent continuously at the 4/5 minute rate. 

I cannot think of reasons why a Resp might not have been received by the CM4 core for any telemetry send Cmd.  

**How to reproduce the bug (skip if none)**

At this time, I cannot pinpoint how to reproduce this bug. In my view it happens randomly at different times. Sometimes the system runs for a few days and the bug occurs or sometimes it happens right away.

**Additional context**

I have set up an rtos queue to not bombard the send API with messages. However, my queue gets full when this issue and no messages are sent. 

** Code Snippet ** 
`void MBMUXIF_LoraSendCmd(void)
{
  /* USER CODE BEGIN MBMUXIF_LoraSendCmd_1 */

  /* USER CODE END MBMUXIF_LoraSendCmd_1 */
  if (MBMUX_CommandSnd(FEAT_INFO_LORAWAN_ID) == 0)
  {
    osSemaphoreAcquire(Sem_MbLoRaRespRcv, osWaitForever);
  }
  else
  {
    Error_Handler();
  }
  /* USER CODE BEGIN MBMUXIF_LoraSendCmd_Last */

  /* USER CODE END MBMUXIF_LoraSendCmd_Last */
}`

** Additional Info/questions **
I think by design this system waits forever on this semaphore. If at all a response is not heard back, can we have some retry mechanism or show it as a communication error callback/ retry mechanism of some kind?

RJMSTM, 2024-03-15:
ST Internal Reference: 176222

metaTinker, 2024-04-02:
@RJMSTM any updates on this?

metaTinker, 2024-05-09:
@ALABSTM @RJMSTM is there any resolution to this?

ALABSTM, 2024-05-27:
Hi @metaTinker,

We got the point. We will get back to you when we have updates to share. This may take some time. Thank you for your comprehension.

With regards,

RJMSTM, 2025-01-30:
Hello @metaTinker,

I hope you are fine. The issue you reported has been fixed in the frame of version v1.3.1 of the STM32CubeWL published recently on GitHub. Thank you again for having reported.

Regards,

closed 2025-01-30 by RJMSTM