STMicroelectronics/STM32CubeWL #81: STM32CubeWL Lorawan_End_node errors

natgeolemme, 2024-03-18:
Dear Sirs,
for a few weeks I have been trying to make this project work with my seonji board (LSM110A) but I couldn't find a way to do it.
I checked the board with the STM32WLxx Wireless Long Range RF test and it works properly. But when I try to use the board with your Lorawan_End_Node example, I always get an error message in the compiler.
FYI, I am using the STM32CUBE IDE 1.14.1
These are the error codes I get:
*stm32wlxx.h: No such file or directory ( I download manually and another error appears).
*radio.o Error 1
*radio_driver.o Error 1
*radio_fw.o Error 1
I will really appreciate any help and I beg your pardon if it's a silly question, I am brand new with all of these.
Thank you in advance
Best regards
Ignacio

ALABSTM, 2024-03-21:
Hi @natgeolemme,

Glad to read your post. All questions are more than welcome to this support channel. We well have a deeper look and try to get back to you soon enough hopefully.

With regards,

RJMSTM, 2024-03-22:
Hello @natgeolemme,

Could you please first try to download the package using this command first:
 `git clone --recursive https://github.com/STMicroelectronics/STM32CubeWL.git` To clone this repository along with the linked submodules.

Second can you try with STM32CubeIDE 1.11.0 version as mentioned in the [Release_Notes](https://htmlpreview.github.io/?https://github.com/STMicroelectronics/STM32CubeWL/blob/master/Release_Notes.html)

Regards,

RJMSTM, 2024-04-02:
Hello,
Any update please !

RJMSTM, 2024-04-05:
Hello @natgeolemme,

Please allow me to close this issue as no activity. You may reopen it at any time if you have any details to share with us in order to help you to solve the issue. Thank you for your comprehension.

With regards