STMicroelectronics/STM32CubeWL #85: [LoRaMac] The use of the adrBlockFound variable in ProcessMacCommands is unknown

wdfk-prog, 2024-05-09:
**Additional context**

- I don't understand what `adrBlockFound` does here, because every time it comes in it's a `false` assignment; A value of `true` below does nothing.

- I think this variable needs to be set to static to make sense.

https://github.com/STMicroelectronics/STM32CubeWL/blob/eec7ff9694007fda9aae32bdb2304c67c0ac013f/Middlewares/Third_Party/LoRaWAN/Mac/LoRaMac.c#L2676-L2774

RJMSTM, 2024-05-13:
ST Internal Reference: 181246

RJMSTM, 2025-01-30:
Hello,

We verified the use of this variable, and it is correctly defined.
It must be set to false at the beginning of the function to process only the first block of LinkAdrRequests and ignore the subsequent blocks as the comment specified:

// The end node is allowed to process one block of LinkAdrRequests.
// It must ignore subsequent blocks

This part is within a While loop to process the all MAC command, so it has effect on the MAC command.

If it were defined static, it will be always true after the first function call, breaking the right functionality.

Regards,
Rania

wdfk-prog, 2025-02-09:
Thanks for explaining. I have no more questions.

closed 2025-02-09 by wdfk-prog