STMicroelectronics/STM32CubeWL #88: SFR and SFD files are missing in Keil pack

MaxFlatline, 2024-07-23:
Hello, since the keil pack is included in this repository, i thought it's ok to ask here. 

**Set-up**

 * Nucleo-WL55JC1
 * Keil Uvision 5.40, STM32WLxx_DFP.1.2.0

**Describe the bug (skip if none)**
No System Viewer fields while debugging CM0P, so i can check periphery only while using CM4

**Additional context**

Seems like something is wrong with provided files 

**Screenshots**
[image]
[image]

RJMSTM, 2024-07-24:
Hello @MaxFlatline,

Thank you for your contribution, but I am unable to reproduce the problem. I conducted a test with

- Nucleo-WL55JC1
- Keil Uvision 5.38, STM32WLxx_DFP.1.2.0

[image]

And everything works well as follows. I am able to see the System Viewer fields.

[image]

Could you please double-check on your side?

Regards,

MaxFlatline, 2024-07-24:
Hello @RJMSTM 
Yes, i checked on another machine and seems like all is operational.. Extremely strange, i'll close this issue and will try to find what's wrong on my side!

Thank you!

MaxFlatline, 2024-07-24:
On another machine all is good, will look into it

closed 2024-08-08 by ALABSTM