STMicroelectronics/STM32CubeWL #89: Device crashes when uplink send after switch to class B.

ankit-thealchemist, 2024-07-24:
**Describe the set-up**

 * The board is NUCLEO-WL55JC1
 * STM32 CubeIDE v1.15.1 
 * STM32 CubeMx v6.10

**Describe the bug (skip if none)**

I am running the Lorawan AT Slave example(although it appears on lorawan end node example also). I configure the lorawan version 1.0.4 with lorawan package 2.0.0(other details in the screenshot). After switiching to class B, when ever I do the uplink, the device crashes. 

**How to reproduce the bug (skip if none)**

1. List the modules that you suspect to be the cause of the problem LoraMacClassB.  After swtiching to the class B device tries to calculate the multicast ping slot offset. Since the multicast setup have not been setup, the default value of the ping period is 0.  In the first state, ie PINGSLOT_STATE_CALC_PING_OFFSET, the device try to calculate the ping offset, since the ping period is 0, the divide by 0 fault crashes the device. 

2. How we can reproduce the problem

- Use the Lorawan AT example 
- Send the command, AT+JOIN=1
- After join, issue command AT+CLASS = B
- wait for the device , switch to class B
- issue command AT+SEND=43:1:112233

**Additional context**

Device should not look for the multicast Class B pingslot until and unless multicast session has been setup.

**Screenshots**

[image]
[image]
[image]

RJMSTM, 2024-07-25:
ST Internal Reference: 187438

pd-vt, 2024-08-13:
We are able to fix this issue as workaround with commenting an If condition as attached image in LoRaMacClassB.c file and LoRaMacClassBProcess() function. 

[image]

We know this is not proper solution but since we didn't had need of Multicast Class-B feature we could bypass this.

closed 2025-09-24 by RJMSTM