STMicroelectronics/STM32CubeWL #95: Device not wake up after RTC-time and send message with Lora P2P.

Leinad1514, 2024-10-25:
I would like to use my ST32 to send data from one location to another. For this purpose, it should sleep in Stop Mode 2 for a fixed period, then wake up, send a message via LoRa P2P, and go back to sleep for this fixed period. This setup is necessary because there is no power supply at the specified location, so the controller needs to be battery-operated for an extended time.

My problem is that the RTC is exclusively reserved for the LoRa function, so I can either put the microcontroller to sleep every 10 seconds without using LoRa or send LoRa messages. When I implement both in the program, the microcontroller goes to sleep, wakes up after the defined period, and then does nothing further — no LoRa message, no returning to sleep for the set duration.

I have already tried initializing the RTC-Clock in two different configurations, but neither was successful.

Can anyone advise me on how to solve this problem? Is it even possible to achieve this with the RTC-Clock, or are there other ways to address my requirements?
[image]
[image]

ALABSTM, 2024-10-29:
Hi @Leinad1514,

According to page **241** of the reference manual [RM0453](https://www.st.com/resource/en/reference_manual/rm0453-stm32wl5x-advanced-armbased-32bit-mcus-with-subghz-radio-solution-stmicroelectronics.pdf), rev. 6, once the `STOP2` mode entered, both the `RTC` and the `sub-GHz radio` can remain active **optionally**.

For the `RTC`, for instance, **table 212** at page **1007** indicates that its clock source **must** be either the `LSI` or the `LSE` clock. The same goes for the `sub-GHz radio`, in the sense that there must certainly be some conditions related to the clock source or another configuration. 

Would you mind checking whether the configurations you are using fulfill these conditions?

With regards,

Leinad1514, 2024-11-04:
[image]
[image]

Here my ClockConfig and RTCConfig. For the RTC I have the LSE Clock.

Leinad1514, 2024-11-15:
Unfortunately my problem still persists. Do you need any further information? Or do you already have an idea what it could be? @ALABSTM

ALABSTM, 2024-11-20:
Hi @Leinad1514,

If I am correct, the wake-up from the `STOP 2` mode causes the program execution to **restart** from the beginning. On the other hand, I see you chose the **RTC Alarm** as a wake-up source. Please note this is a **one shot** wake-up source.

Hence, you should ensure your code structure makes it possible for the RTC Alarm to be **rearmed** after each wake-up and program reset.

Moreover, I also see you configured your alarm this way `sAlarm.AlarmTime.SubSeconds = 0x0;`. I am not sure this is a good initialization value. Below a suggestion you may find more suitable:

https://github.com/STMicroelectronics/STM32CubeWL/blob/99e6b06c87338a6d78a7454e8b455b119cc29ac8/Projects/NUCLEO-WL55JC/Examples/RTC/RTC_Binary/Src/main.c#L233

You can have a look at [this](https://github.com/STMicroelectronics/STM32CubeWL/tree/main/Projects/NUCLEO-WL55JC/Examples/RTC/RTC_Binary) example to support you.

----

Now, I would also suggest you rather choose the **RTC Wake-Up Timer** as a **recurrent** wake-up source. You can have a look at [this](https://github.com/STMicroelectronics/STM32CubeWL/tree/main/Projects/NUCLEO-WL55JC/Examples/PWR/PWR_STOP2_RTC) example.

I hope this helps. Please let me know.

With regards,

ALABSTM, 2024-11-26:
Hi @Leinad1514,

I hope my previous comment is helpful. If your issue still persists, I recommend that you submit a post to the [ST Community](https://community.st.com/t5/stm32-mcus/ct-p/stm32-mcus).

Please allow me to close this thread. Thank you for your comprehension and good luck.

With regards,

closed 2024-11-26 by ALABSTM