STMicroelectronics/STM32CubeWL #96: why this code is not supporting LoRa E5 Module (custom board) ? Anything we need to change the pinout configuration ?

Manjunath2001-debug, 2025-01-02:
**Caution**

The Issues are strictly limited for the reporting of problem encountered with the software provided in this project.
For any other problem related to the STM32 product, the performance, the hardware characteristics and boards, the tools the environment in general, please post your report to the **ST Community** in the STM32 MCUs dedicated [page](https://community.st.com/s/topic/0TO0X000000BSqSWAW/stm32-mcus).

**Describe the set-up**

 * The board (either ST RPN reference or your custom board)
 * IDE or at least the compiler and its version

**Describe the bug (skip if none)**

A clear and concise description of what the bug is.

**How to reproduce the bug (skip if none)**

1. Indicate the global behavior of your application project
2. List the modules that you suspect to be the cause of the problem (Drivers, BSP, MW...)
3. Describe the use case that generates the problem
4. How we can reproduce the problem

**Additional context**

If you have a first analysis, an enhancement, a fix or a patch, thank you to share your proposal.

**Screenshots**

If applicable, add screenshots to help explain your problem.

Manjunath2001-debug, 2025-01-06:
LoRawan_end_node example is not supporting to LoRa-E5(STM32WLE5JC) Module ?

Can you please provide some idea, where we need change the code in this stack ?

Thank you in advance

RJMSTM, 2025-01-31:
Hello @Manjunath2001-debug,

The STM32WLE5JC is the single-core variant of the WL55.

Since the end-node application is designed to operate on a single core, it should function correctly, however, we have not yet tested it on the WLE5 board.

Regarding class B: there is some support in the code, but it is not fully supported.

Regards,

RJMSTM, 2025-02-05:
Hello @Manjunath2001-debug,

If the response is satisfactory for you, I will close the ticket. Otherwise, if you have another question, you can reopen the issue. Thank you once again for your contribution.

Regards,

closed 2025-02-05 by RJMSTM